micropython/micropython #17081: RP2040 Cannot plug gap greater than alignment

edtink, 2025-04-05:
### Port, board and/or hardware

Pico (RP2040, no Wifi)

### MicroPython version

MicroPython master branch (04/04/24), Pico board, RP5 environment with latest upgrades.

Clean build following instructions "1.3. Building MicroPython From Source" in "https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-python-sdk.pdf" fails to link with error:

[ 98%] Linking CXX executable firmware.elf
ERROR: Cannot plug gap greater than alignment - gap 254000, alignment 32

Clean build with branch v1.24-release builds successfully. Note that I have used the master branch successfully in the recent past (a few weeks ago).

### Reproduction

Basic build following instructions "1.3. Building MicroPython From Source" in "https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-python-sdk.pdf"

### Observed behaviour

Build fails to link with error:

[ 98%] Linking CXX executable firmware.elf
ERROR: Cannot plug gap greater than alignment - gap 254000, alignment 32

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

dlech, 2025-04-05:
Duplicate of #17080?

nkrba, 2025-04-05:
best resolution is as mentioned by dpgeorge in https://github.com/micropython/micropython/pull/17073
edit `GIT_TAG develop` to `GIT_TAG master` within `pico-sdk/tools/Findpicotool.cmake`

closed 2025-04-05 by dpgeorge