micropython/micropython #17093: mpremote mkdir crashes on Windows if path does not start with :

Josverl, 2025-04-08:
### Port, board and/or hardware

Any, verified on esp32 and stm32 

### MicroPython version

mpflash list --json

[
    {
        "serialport": "COM24",
        "connected": true,
        "family": "micropython",
        "description": "PYBv1.1 with STM32F405RG",
        "version": "1.25.0-preview",
        "port": "stm32",
        "board": "PYBV11",
        "variant": "DP_THREAD",
        "cpu": "STM32F405RG",
        "arch": "armv7emsp",
        "mpy": "v6.3",
        "build": "389",
        "location": "1-6.1.1:x.1",
        "toml": {}
    }
]

### Reproduction

The mip local install crashes in **Git Bash** on Windows 

```
mpremote run "./ramdisk.py"
mpremote resume mkdir /ramdisk/lib
```
Note that this problem does NOT occur on Windows Powershell , it is Git Bash specific.

(this also means that Git bash is not a reliable tests environment for windows as a whole.) 

### Expected behaviour

Expected to create a folder in the ramdisk

### Observed behaviour

```sh
EUROPE+josverl@josverl-sb5 MINGW64 /d/mypython/micropython/tools/mpremote/tests (mpremote-testfix)
$ mpremote run "./ramdisk.py"
mpremote resume mkdir /ramdisk/lib
mkdir :C:/Program Files/Git/ramdisk/lib
mpremote: Error with transport:
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
OSError: [Errno 22] EINVAL

mpremote resume mkdir :/ramdisk/lib
mkdir :/ramdisk/lib
```

### Additional Information

discovered during attempts to test mpremote on windows.

### Code of Conduct

Yes, I agree

AJMansfield, 2025-04-08:
Could you rerun your scenario with `set -x` tracing enabled and post the results?

Josverl, 2025-04-09:
just for clarity: This report is for the error in mpremote, not regarding the bash test harness.
So the bash tracing does not show much 
I think the root cause is Git Bash 

the workaround is simple: use a `:` prefix for `mpremote mkdir :</absolute/path>`

I Suggest to: 
- [ ] Add this to the mpremote docs 
- [ ] update the [mpremote test](https://github.com/micropython/micropython/blob/ef8282c717be8363e9a04ebf2b9d843e2485604f/tools/mpremote/tests/test_mip_local_install.sh#L60) to use a `:` path prefix , 
- [ ] or handle the error as XFAIL on git bash ( requires a different test framework)

if I set a breakpoint on entry and look at the arguments passed to Python 
For a command line of : `mpremote resume mkdir /ramdisk/lib`
``` 
sys.argv
['D:\\mypython\\micropython\\tools\\mpremote\\mpremote\\__main__.py', 'resume', 'mkdir', 'C:/Program Files/Git/ramdisk/lib']
special variables
function variables
0 ='D:\\mypython\\micropython\\tools\\mpremote\\mpremote\\__main__.py'
1 = 'resume'
2 ='mkdir'
3 ='C:/Program Files/Git/ramdisk/lib'
len() =4
```
I syspect that this is a (unexpected) side effect ot the way Git bash tries to translate absolute paths between the Posix universe with a single tree , and the Windows universe that allows both drive letters , and mount points.
```bash
#Git Bash
$ mount
C:/Program Files/Git on / type ntfs (binary,noacl,auto)
C:/Program Files/Git/usr/bin on /bin type ntfs (binary,noacl,auto)
C:/Users/josverl/AppData/Local/Temp on /tmp type ntfs (binary,noacl,posix=0,usertemp)
C: on /c type ntfs (binary,noacl,posix=0,user,noumount,auto)
D: on /d type refs (binary,noacl,posix=0,user,noumount,auto)
E: on /e type ntfs (binary,noacl,posix=0,user,noumount,auto)
F: on /f type vfat (binary,noacl,posix=0,user,noumount,auto)
```

<details><summary>bash Details</summary>
<p>

likely not relevant as explained above
```
$ set -x; mpremote resume mkdir /ramdisk/lib;set +x
++ __vsc_preexec_only -x
++ '[' 1 = 0 ']'
+ mpremote resume mkdir /ramdisk/lib
mkdir :C:/Program Files/Git/ramdisk/lib
mpremote: Error with transport:
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
OSError: [Errno 22] EINVAL

++ __vsc_preexec_only /ramdisk/lib
++ '[' 1 = 0 ']'
+ set +x
```

</p>
</details>

AJMansfield, 2025-04-09:
This is a bizarre bug. I've checked, this also happens with [jq](https://jqlang.org/) but NOT with the utilities that ship with Git Bash.
It also doesn't happen with other windows bash shells, e.g. the one included in the [GAP](https://www.gap-system.org/) runtime doesn't have this problem at all.

Ordinary quoting doesn't seem to prevent it -- `<command> '/ramdisk/lib'` is still expanded, and so is `<command> "/ramdisk/lib"`, however `<command> \'/ramdisk/lib\'` (or the expansion generated by `@Q`) does -- but using this quoting does _not_ work in on linux.

<details>
<summary>Test Script</summary>

```bash
#!/bin/bash
set -x
cat <<<'==================== Test Script ===================='
TEST_SH=$(mktemp --suffix=.sh)
tee $TEST_SH <<'TEST_SH'
set -x
f=/ramdisk/lib

ECHO_EXE=$(mktemp --suffix=.exe)
cp $(which echo) $ECHO_EXE

GREP_TXT=$(mktemp --suffix=.txt)
tee $GREP_TXT <<GREP_TXT
C:/Program Files/Git$f
did not expand $f if this line appears
GREP_TXT

ARGS_PY=$(mktemp --suffix=.py)
tee $ARGS_PY <<ARGS_PY
import sys
print(sys.argv)
ARGS_PY

cat <<<'--------------- echo: shell builtin? ----------------'
echo $f
echo ${f@Q}
echo \'$f\'

cat <<<'------------------ echo: echo.exe -------------------'
$ECHO_EXE $f
$ECHO_EXE ${f@Q}
$ECHO_EXE \'$f\'

cat <<<'----------- grep: winget install Git.Git ------------'
grep -F $f <$GREP_TXT
grep -F ${f@Q} <$GREP_TXT
grep -F \'$f\' <$GREP_TXT

cat <<<'---- python: winget install Anaconda.Miniconda3 -----'
python $ARG_PY -- $f
python $ARG_PY -- ${f@Q}
python $ARG_PY -- \'$f\'

cat <<<'----------- jq: winget install jqlang.jq ------------'
jq '{"file":input_filename}' $f
jq '{"file":input_filename}' ${f@Q}
jq '{"file":input_filename}' \'$f\'
TEST_SH

cat <<<'=================== Git /bin/bash ==================='
'C:/Program Files/Git/bin/bash.exe' $TEST_SH

cat <<<'================= Git /usr/bin/bash ================='
'C:/Program Files/Git/usr/bin/bash.exe' $TEST_SH

cat <<<'=============== GAP Runtime /bin/bash ==============='
"C:/Program Files/GAP-v4.12.2/runtime/bin/bash.exe" $TEST_SH
```

</details>

[... 203 lines cut ...]
did not expand /ramdisk/lib if this line appears
+ grep -F ''\''/ramdisk/lib'\'''
C:/Program Files/Git/ramdisk/lib
did not expand /ramdisk/lib if this line appears
+ grep -F ''\''/ramdisk/lib'\'''
C:/Program Files/Git/ramdisk/lib
did not expand /ramdisk/lib if this line appears
+ cat
---- python: winget install Anaconda.Miniconda3 -----
+ python -- /ramdisk/lib
C:\Users\anson\anaconda3\envs\py311\python.exe: can't open file 'C:\\ramdisk\\lib': [Errno 2] No such file or directory
+ python -- ''\''/ramdisk/lib'\'''
C:\Users\anson\anaconda3\envs\py311\python.exe: can't open file "C:\\Users\\anson\\'\\ramdisk\\lib'": [Errno 2] No such file or directory
+ python -- ''\''/ramdisk/lib'\'''
C:\Users\anson\anaconda3\envs\py311\python.exe: can't open file "C:\\Users\\anson\\'\\ramdisk\\lib'": [Errno 2] No such file or directory
+ cat
----------- jq: winget install jqlang.jq ------------
+ jq '{"file":input_filename}' /ramdisk/lib
jq: error: Could not open file /ramdisk/lib: No such file or directory
+ jq '{"file":input_filename}' ''\''/ramdisk/lib'\'''
jq: error: Could not open file '/ramdisk/lib': No such file or directory
+ jq '{"file":input_filename}' ''\''/ramdisk/lib'\'''
jq: error: Could not open file '/ramdisk/lib': No such file or directory
```
</details>

AJMansfield, 2025-04-09:
Another interesting find -- deactivating my conda environment fixed this.

Josverl, 2025-04-10:
Please relabel as [Docs] [Tests]

AJMansfield, 2025-04-11:
> Possibly in https://github.com/msys2/MSYS2-packages/tree/master/bash but not something we can solve here.

It's _possible_ there's something we could do about it for our own package. I'm currently looking into how it is that the programs that ship as built-in MSYS2 utilities are able to accept their path-like arguments untransformed -- since, one way or another, they definitely _do_ receive their arguments without the transformation that Git Bash applies when running other programs like python and jq.

AJMansfield, 2025-04-11:
Found a page that documents this feature of MSYS2:
https://www.msys2.org/docs/filesystem-paths/#windows-unix-path-conversion

Josverl, 2025-04-14:
I don't have time to test this now, but plan to look at this next week

Josverl, 2025-05-09:
Finally got around to testing this.
Your workaround indeed works, but I think its better to document this than implement a script to around this.