micropython/micropython #17097: micropython.kbd_intr(-1) not working on ESP32*

xuancong84, 2025-04-09:
### Port, board and/or hardware

esp32, esp32c3, esp32*

### MicroPython version

MicroPython v1.24.1 on 2024-11-29; ESP32C3 module with ESP32C3
Type "help()" for more information.

### Reproduction

Disabling CTRL+C interrupt is very important and useful because it allows STDIO ports to be used as UART ports for interaction with external devices. On ESP32*, REPL cannot be detached by `os.dupterm(None,...)`, we have to use `sys.stdin.buffer` (which is the same as `sys.stdout.buffer`) as UART to communicate with other devices.

Under such a scenario, the transmitted data can contain arbitrary bytes including b'\x03' which means CTRL+C and it should not interrupt the execution of the program. `micropython.kbd_intr(-1)` is designed for such a purpose. However, it does not work on the latest MicroPython builds for ESP32*:
```
>>> micropython.kbd_intr(-1)         
>>> while True:pass                  
...                                  
Traceback (most recent call last):   
  File "<stdin>", line 1, in <module>
KeyboardInterrupt:                   
```
No matter whether the code is run in REPL terminal or not, CTRL+C will always interrupt the execution. The reason is because in `parse_compile_execute()` in `pyexec.c`, every time it will reset `interrupt_char` to CHAR_CTRL_C:
```
if (!(exec_flags & EXEC_FLAG_NO_INTERRUPT)) {
    mp_hal_set_interrupt_char(CHAR_CTRL_C);
}
#if MICROPY_REPL_INFO
start = mp_hal_ticks_ms();
#endif
mp_call_function_0(module_fun);
mp_hal_set_interrupt_char(-1); // disable interrupt
mp_handle_pending(true); // handle any pending exceptions (and any callbacks)
nlr_pop();
ret = 1;
if (exec_flags & EXEC_FLAG_PRINT_EOF) {
    mp_hal_stdout_tx_strn("\x04", 1);
}
```

### Expected behaviour

```
>>> micropython.kbd_intr(-1)         
>>> while True:pass                  
...            
```
By right, pressing CTRL+C should not interrupt the execution, it should stuck forever.

### Observed behaviour

```
>>> micropython.kbd_intr(-1)         
>>> while True:pass                  
...                                  
Traceback (most recent call last):   
  File "<stdin>", line 1, in <module>
KeyboardInterrupt:                   
```

### Additional Information

My commit https://github.com/xuancong84/micropython/commit/2e7cd9f392c3771bfe4e9646faef630bc618fbfa resolves this issue by adding `micropython.kbd_intr_enable(bool)`. However, my understanding of the full source code is quite limited, there might be a more sophisticated solution.

### Code of Conduct

Yes, I agree

agatti, 2025-04-09:
I don't have a working ESP32 environment on this machine right now to test this but wouldn't changing

```c
if (!(exec_flags & EXEC_FLAG_NO_INTERRUPT)) {
    mp_hal_set_interrupt_char(CHAR_CTRL_C);
}
```

into

```c
if (!(exec_flags & EXEC_FLAG_NO_INTERRUPT) && (mp_interrupt_char != - 1)) {
    mp_hal_set_interrupt_char(CHAR_CTRL_C);
}
```

make it work like you want it to?

robert-hh, 2025-04-09:
The whole thing looks like a non-issue. Indeed the keyboard interrupt char is reset to Ctrl-C at every entered on the  command line in REPL. That is the problem with the above example. But if a Python program disables Ctrl-C interrupt, it stays disabled until that program finishes and enters REPL again. I have code which disables Ctrl_C as interrupt char and uses sys.stdin.read() to input Ctr-C like any other code. And that works identical at all ports. No change is needed for pyexec,c.

xuancong84, 2025-04-10:
@agatti I saw `mp_hal_set_interrupt_char(-1)` somewhere down that line. It seems that the code will set it to -1 to disable interrupt so it will cause confusion whether it is done by code or done by user program. Actually, I have tested that after `micropython.kbd_intr(-1)`, although REPL CTRL+C will interrupt execution, UART input will no longer interrupt execution. So no change is needed for practice. Thanks!

closed 2025-04-10 by xuancong84