micropython/micropython #17110: Support for parenthesized `with` statements

smurfix, 2025-04-10:
### Description

Python 3.9 (I think) introduced multi-line `with` statements along the lines of
```python
        async with (
            Foo() as bar,
            Baz() as quux,
        ):
            ...
```
Please support.

### Implementation

I would like to sponsor development of this feature

### Code of Conduct

Yes, I agree

dpgeorge, 2025-04-10:
Is this the same as the existing:
```py
[async] with Foo() as bar, Baz() as quux:
    ...
```
?

If so then it's just a slight modification of the syntax to allow parenthesis?

Can you point to the PEP or issue/discussion/PR on CPython where this was implemented?

dlech, 2025-04-10:
It was [Python 3.10](https://docs.python.org/3/whatsnew/3.10.html). 

> [bpo-12782](https://bugs.python.org/issue?@action=redirect&bpo=12782), Parenthesized context managers are now officially allowed.

smurfix, 2025-04-10:
> Is this the same as

Yes.

> If so then it's just a slight modification of the syntax to allow parenthesis?

Should be, but I'd like to leave that change to people who don't need to spend as much time understanding the µPy parser than I'd need to implement that. *sigh*

dpgeorge, 2025-04-10:
OK, thanks for the details.

So it is just a "minor" syntax enhancement.  Although reading the bpo-12782 it sounds like it was not an easy thing to add to the CPython grammar... I'll have to see how we could add it.

sosi-deadeye, 2025-04-12:
I am not completely sure. I think the change was possible with the new parser: https://peps.python.org/pep-0617/

```
[deadeye@nexus ~]$ python
Python 3.8.19 (default, Jun 30 2024, 11:46:07)
[GCC 14.1.1 20240522] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import contextlib
>>> with (contextlib.nullcontext() as a, contextlib.nullcontext() as b):
  File "<stdin>", line 1
    with (contextlib.nullcontext() as a, contextlib.nullcontext() as b):
                                   ^
SyntaxError: invalid syntax
>>>
```

```
[deadeye@nexus ~]$ pyenv shell 3.9.22
[deadeye@nexus ~]$ python
Python 3.9.22 (main, Apr  9 2025, 21:44:38)
[GCC 14.2.1 20250207] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import contextlib
>>> with (contextlib.nullcontext() as a, contextlib.nullcontext() as b):
...     ...
...
Ellipsis
>>>

```

Before the change, it was only possible to do it without parenthesis.

```
[deadeye@nexus ~]$ pyenv shell 3.8.19
[deadeye@nexus ~]$ python
Python 3.8.19 (default, Jun 30 2024, 11:46:07)
[GCC 14.1.1 20240522] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import contextlib
>>> with contextlib.nullcontext() as a, contextlib.nullcontext() as b:
...     ...
...
Ellipsis
>>>
```

dpgeorge, 2025-04-12:
> I think the change was possible with the new parser:

Yes, you are right, CPython needed the PEG parser before it could support parenthesis around the with statements.

So, the question is whether or not MicroPython is able to implement this with its current parser.