micropython/micropython #17126: Neopixel "flicker"/noise only when Wifi is on

jomasnash, 2025-04-14:
### Port, board and/or hardware

esp32-C3 Mini

### MicroPython version

MicroPython v1.24.1 on 2024-11-29; ESP32C3 module with ESP32C3

### Reproduction

When Wifi is enabled and connected, a neopixel matrix (8x8) gives unexpected glitches.
Power and signal levels are checked and ok. (checked with an oscilloscope.)

This bug is the same as in https://github.com/micropython/micropython/issues/8161 however the fix will not work because the esp32-C3 Mini has only one core.

If this bug cannot be solved, I suggest adding a warning in the documentation that eps32 (single core types) cannot be used reliable when Wifi is on and used. 

### Observed behaviour

Glitches appear

### Additional Information

 

### Code of Conduct

Yes, I agree

dpgeorge, 2025-04-15:
Can you please give more information so the issue can be reproduced:
- neopixel matrix used (exact model of the neopixel/WS chip)
- neopixel driver used
- code to run that shows the issue

smb47-null, 2025-04-15:
```shell

``` /`

smb47-null, 2025-04-15:
Hey

ricksorensen, 2025-04-15:
There is some talk about this in various esp32 sites and the two core solution seems prevelant but that does not help for C3.  

Here is a PR that uses the IDF 5x RMT library - #16293.  It may be worth a look, though I think you need IDF v5.4 for it.

jomasnash, 2025-04-15:
> Can you please give more information so the issue can be reproduced:
> 
> * neopixel matrix used (exact model of the neopixel/WS chip)
> * neopixel driver used
> * code to run that shows the issue

See updates in start of topic