micropython/micropython #17127: Unable to setup IP with nic.ipconfig() while works with obsolete nic.ifconfig() network.WIZNET5K

matkor, 2025-04-15:
### Port, board and/or hardware

RPi Pico, W5500-EVB-Pico

### MicroPython version

MicroPython v1.24.1 on 2024-11-29; W5500-EVB-Pico with RP2040

### Reproduction

Run below code with
`# nic.ipconfig(addr4="192.168.9.12/24",gw4="192.168.9.3")  # No pings reply`
nor
`# nic.ipconfig(addr4=("192.168.9.12","255.255.255.0"),gw4="192.168.9.3")  # No ping reply`
```
import network
import time

nic = network.WIZNET5K()
print (nic)
nic.active(True)
print (f"Activated {nic=}")

# nic.ifconfig(('192.168.9.12', '255.255.255.0', '192.168.9.3', '192.168.9.3')) # Pings OK
nic.ipconfig(addr4="192.168.9.12/24",gw4="192.168.9.3")  # No pings reply
# nic.ipconfig(addr4=("192.168.9.12","255.255.255.0"),gw4="192.168.9.3")  # No ping reply
network.ipconfig(dns="192.168.9.3", prefer=4)

print(f'After static nic config {nic=} {nic.ifconfig()=}')
print(f'After static network config {network.ipconfig("dns")=} {network.ipconfig("prefer")=}')
# In all three cases prints same:
# After static nic config nic=<WIZNET5K> nic.ifconfig()=('192.168.9.12', '255.255.255.0', '192.168.9.3', '192.168.9.3')
# After static network config network.ipconfig("dns")=192.168.9.3 network.ipconfig("prefer")=4

# nic.connect()  AttributeError: 'WIZNET5K' object has no attribute 'connect'
print("Waiting for connection...")
while not nic.isconnected():
    print ("Not connected")
    time.sleep(1)
print ("Connected")
while 1:
    time.sleep(3)
    print (f"{nic.isconnected()=}")
```

### Expected behaviour

Reply to ICMP ping when NIC configured with `nic.ipconfig()` 

### Observed behaviour

No reply to ICMP Ping when configured with `nic.ipconfig()` 
But ICMP Ping replies received after init via `nic.ifconfig()`.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

MilhouseVH, 2025-04-16:
Yeah, something strange is happening.

Running your test code on a W5500_EVB_PICO_POE results in it executing a DHCP request - it's ignoring your static network configuration.

This is the output using your unmodified test code:
```
MPY: soft reboot
<WIZNET5K>
Activated nic=<WIZNET5K>
After static nic config nic=<WIZNET5K> nic.ifconfig()=('192.168.9.12', '255.255.255.0', '192.168.9.3', '192.168.9.3')
After static network config network.ipconfig("dns")=192.168.9.3 network.ipconfig("prefer")=4
Waiting for connection...
Not connected
Not connected
Connected
nic.isconnected()=True
nic.isconnected()=True
nic.isconnected()=True
...
```

My DHCP server (`dnsmasq` on a RPi5 with IP `192.168.0.4`) sees the DHCP request from the W5500_EVB_PICO_POE and issues IP address `192.168.0.226` (and probably gateway `192.168.0.1`, with DNS `192.168.0.4`):
```
Apr 16 01:13:22 dnsmasq-dhcp[740]: DHCPDISCOVER(eth0) 02:9c:17:50:63:9a
Apr 16 01:13:22 dnsmasq-dhcp[740]: DHCPOFFER(eth0) 192.168.0.226 02:9c:17:50:63:9a
Apr 16 01:13:22 dnsmasq-dhcp[740]: DHCPREQUEST(eth0) 192.168.0.226 02:9c:17:50:63:9a
Apr 16 01:13:22 dnsmasq-dhcp[740]: DHCPACK(eth0) 192.168.0.226 02:9c:17:50:63:9a
```
My W5500_EVB_PICO_POE will now respond to pings on `192.168.0.226`, so my guess is that your device is being issued another address on `192.168.9.0/24`, and that it's not actually using `192.168.9.12` at all.

So the question is why does `ipconfig()` not configure the interface using the specified parameters... even though it responds as though those parameters have been set? Is DHCP some kind of fallback?

I'm currently using `ifconfig()` in my own project (which as you've stated, works as expected - the specified addresses are configured) but converting to use `ipconfig()` with micropython 1.24.1 results in the same behaviour that you have reported here - my device picks up IP `192.168.0.226` via DHCP even though I've set it to use `192.168.0.227/24` etc.

MilhouseVH, 2025-04-16:
I'm going to go out on a limb and suggest that `network.WIZNET5K().ipconfig()` is not actually implemented... or not implemented correctly... because so long as you call (at minimum) `network.ipconfig(dns="192.168.9.3", prefer=4)` then you'll end up with a DHCP assigned IP address regardless, as this appears to be the default behaviour.

The following test code with just the DNS call to `network.ipconfig()` will result in a DHCP-assigned IP address (`192.168.0.226` in my case) - I've also output the configuration once the connection is established to reveal the true network configuration that is being used:
```
import network
import time

nic = network.WIZNET5K()
print (nic)
nic.active(True)
print (f"Activated {nic=}")

# nic.ifconfig(('192.168.9.12', '255.255.255.0', '192.168.9.3', '192.168.9.3')) # Pings OK
#nic.ipconfig(addr4="192.168.9.12/24",gw4="192.168.9.3")  # No pings reply
# nic.ipconfig(addr4=("192.168.9.12","255.255.255.0"),gw4="192.168.9.3")  # No ping reply
network.ipconfig(dns="192.168.9.3", prefer=4)

print(f'After static nic config {nic=} {nic.ifconfig()=}')
print(f'After static network config {network.ipconfig("dns")=} {network.ipconfig("prefer")=}')
# In all three cases prints same:
# After static nic config nic=<WIZNET5K> nic.ifconfig()=('192.168.9.12', '255.255.255.0', '192.168.9.3', '192.168.9.3')
# After static network config network.ipconfig("dns")=192.168.9.3 network.ipconfig("prefer")=4

# nic.connect()  AttributeError: 'WIZNET5K' object has no attribute 'connect'
print("Waiting for connection...")
while not nic.isconnected():
    print ("Not connected")
    time.sleep(1)
print ("Connected")
while 1:
    time.sleep(3)
    print (f"{nic.isconnected()=}")
    print(f'After static nic config {nic=} {nic.ifconfig()=}')
    print(f'After static network config {network.ipconfig("dns")=} {network.ipconfig("prefer")=}')
```
Output:
```
MPY: soft reboot
<WIZNET5K>
Activated nic=<WIZNET5K>
After static nic config nic=<WIZNET5K> nic.ifconfig()=('0.0.0.0', '0.0.0.0', '0.0.0.0', '192.168.9.3')
After static network config network.ipconfig("dns")=192.168.9.3 network.ipconfig("prefer")=4
Waiting for connection...
Not connected
Not connected
Not connected
Not connected
Not connected
Not connected
Connected
nic.isconnected()=True
After static nic config nic=<WIZNET5K> nic.ifconfig()=('192.168.0.226', '255.255.255.0', '192.168.0.1', '192.168.0.4')
After static network config network.ipconfig("dns")=192.168.0.4 network.ipconfig("prefer")=4
nic.isconnected()=True
After static nic config nic=<WIZNET5K> nic.ifconfig()=('192.168.0.226', '255.255.255.0', '192.168.0.1', '192.168.0.4')
After static network config network.ipconfig("dns")=192.168.0.4 network.ipconfig("prefer")=4
...
```
This version does take consistently longer to connect (6x iterations vs. 2x), so maybe `nic.ipconfig(addr4="192.168.9.12/24",gw4="192.168.9.3")` is doing _something_... at a guess it may be trying to set the IP address (and/or gateway), failing, and then the default DHCP address is eventually requested which overwrites the configured parameters. 🤷 

And using my own `192.168.0/24` subnet produces identical results.

Probably a WIZNET5K specific issue? And also only when using LWIP? 🤷

felixdoerre, 2025-04-20:
The LWIP-specific code for wiznet5k is just calling generic lwip-handling code: https://github.com/micropython/micropython/blob/master/extmod/network_wiznet5k.c#L939-L948

Maybe you want to try to disable dhcp4 actively: `nic.ipconfig(dhcp4=False)`. Currently the code that sets the static ip address does not seem to do that automatically.

MilhouseVH, 2025-04-20:
> Maybe you want to try to disable dhcp4 actively: nic.ipconfig(dhcp4=False). Currently the code that sets the static ip address does not seem to do that automatically.

Yes, that works! So the following calls to `ipconfig()` now work as expected:
```
nic.ipconfig(addr4="192.168.0.227/24",gw4="192.168.0.1",dhcp4=False)  # Pings OK
network.ipconfig(dns="192.168.0.4", prefer=4)
```
and then the correct IP etc. address is configured, and it responds to pings.  👍 
```
MPY: soft reboot
<WIZNET5K>
wiznet5k_send_ethernet: fatal error -5
Activated nic=<WIZNET5K>
After static nic config nic=<WIZNET5K> nic.ifconfig()=('192.168.0.227', '255.255.255.0', '192.168.0.1', '192.168.0.4')
After static network config network.ipconfig("dns")=192.168.0.4 network.ipconfig("prefer")=4
Waiting for connection...
Not connected
Not connected
Connected
nic.isconnected()=True
After static nic config nic=<WIZNET5K> nic.ifconfig()=('192.168.0.227', '255.255.255.0', '192.168.0.1', '192.168.0.4')
After static network config network.ipconfig("dns")=192.168.0.4 network.ipconfig("prefer")=4
nic.isconnected()=True
After static nic config nic=<WIZNET5K> nic.ifconfig()=('192.168.0.227', '255.255.255.0', '192.168.0.1', '192.168.0.4')
After static network config network.ipconfig("dns")=192.168.0.4 network.ipconfig("prefer")=4
nic.isconnected()=True
After static nic config nic=<WIZNET5K> nic.ifconfig()=('192.168.0.227', '255.255.255.0', '192.168.0.1', '192.168.0.4')
After static network config network.ipconfig("dns")=192.168.0.4 network.ipconfig("prefer")=4
```