micropython/micropython #17128: Request modifies header argument

umpfel, 2025-04-15:
### Port, board and/or hardware

RP2040

### MicroPython version

MicroPython v1.24.1 on 2024-11-29; Raspberry Pi Pico W with RP2040

### Reproduction

Replace ssid / psk and run the code
```
import network
import requests
import time

def do_connect():
    import network
    sta_if = network.WLAN(network.WLAN.IF_STA)
    if not sta_if.isconnected():
        print('connecting to network...')
        sta_if.active(True)
        sta_if.connect("SSID", "PSK")
        while not sta_if.isconnected():
            pass
    print('network config:', sta_if.ipconfig('addr4'))

do_connect()

HEADERS = {"X-Api-Key": "foobar"}
print(HEADERS)
requests.request(method = "GET", url = "https://postman-echo.com/get", headers = HEADERS, timeout=2)
print(HEADERS)
```

### Expected behaviour

Expected output (working with MicroPython v1.23.0 on 2024-06-02; Raspberry Pi Pico W with RP2040)
```
{'X-Api-Key': 'foobar'}
{'X-Api-Key': 'foobar'}
```

### Observed behaviour

```
{'X-Api-Key': 'foobar'}
{'Host': 'postman-echo.com', 'X-Api-Key': 'foobar', 'Connection': 'close'}
```

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

dpgeorge, 2025-04-15:
This should be fixed by https://github.com/micropython/micropython-lib/commit/65a14116d503dd544cbba92284517c99e5e9a448 .  Please retest with latest MicroPython 1.25.0.

umpfel, 2025-04-15:
Yes, it's working with MicroPython v1.25.0-preview.542.g9f3062799 on 2025-04-15; Raspberry Pi Pico W with RP2040
Thanks!

closed 2025-04-15 by umpfel