micropython/micropython #17140: mpremote romfs deploy throws exception

peterhinch, 2025-04-16:
### Port, board and/or hardware

RP2040 (Pi Pico)

### MicroPython version

MicroPython v1.25.0-dirty on 2025-04-16; Raspberry Pi Pico with RP2040 Built from today's source with 128KiB romfs enabled.

Python 3.10.12 (main, Feb  4 2025, 14:57:36) [GCC 11.4.0] on linux

### Reproduction

```bash
$ mpremote romfs deploy fonts.romfs
Image size is 125508 bytes
ROMFS0 partition has size 131072 bytes (32 blocks of 4096 bytes each)
Preparing ROMFS0 partition for writing
Traceback (most recent call last):
  File "/mnt/qnap2/data/Projects/MicroPython/micropython/tools/mpremote/mpremote.py", line 6, in <module>
    sys.exit(main.main())
  File "/mnt/qnap2/data/Projects/MicroPython/micropython/tools/mpremote/mpremote/main.py", line 569, in main
    handler_func(state, args)
  File "/mnt/qnap2/data/Projects/MicroPython/micropython/tools/mpremote/mpremote/commands.py", line 688, in do_romfs
    _do_romfs_deploy(state, args)
  File "/mnt/qnap2/data/Projects/MicroPython/micropython/tools/mpremote/mpremote/commands.py", line 652, in _do_romfs_deploy
    romfs_chunk_compressed = zlib.compress(romfs_chunk, wbits=-9)
TypeError: 'wbits' is an invalid keyword argument for compress()
```
The CPython docs indicate that `wbits` is a positional-only arg

### Expected behaviour

Expected to deploy the romfs file to an RP2 built with romfs enabled.

### Observed behaviour

Threw exception described above.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

BrianPugh, 2025-04-16:
it looks like the `wbits` field was [added in 3.11](https://docs.python.org/3/library/zlib.html#zlib.compress) and is keyword-only.

The more python-version-agnostic way of zlib compressing stuff is using `compressobj`. Something like:

```python
   import zlib

   def compress_data(data):
       w_bits = -9
       level = zlib.Z_BEST_COMPRESSION
       compressor = zlib.compressobj(level=level, method=zlib.DEFLATED,
               wbits=w_bits, memLevel=zlib.DEF_MEM_LEVEL, strategy=zlib.Z_DEFAULT_STRATEGY)
       compressed_data = compressor.compress(data)
       compressed_data += compressor.flush()
       return compressed_data
```

dpgeorge, 2025-04-16:
> The CPython docs indicate that wbits is a positional-only arg

As @BrianPugh points out, the `wbits` parameter was added to `zlib.compress` only in CPython 3.11.  That's a shame 😞 

For backwards compatibility with CPython < 3.11 we should use `zlib.compressobj` instead.  It looks like a value of `wbits=-9` to produce a raw deflate output is supported down to at least CPython 3.5, which is back in time enough.

dpgeorge, 2025-04-17:
See #17143 for a fix.

And thanks @peterhinch for the testing ROMFS and reporting the issue.  It's good to have people using ROMFS so we can iron out all the issues before it gets used more widely.

peterhinch, 2025-04-17:
Now working OK.

closed 2025-04-17 by peterhinch