micropython/micropython #17156: Bytecode stored in romfs appears to be executed in RAM

peterhinch, 2025-04-18:
### Port, board and/or hardware

RP2040 (Pi Pico)

### MicroPython version

MicroPython v1.25.0 on 2025-04-15; Raspberry Pi Pico with RP2040

Also custom builds using V1.25.0 source

### Reproduction

Locate the files in [this directory](https://github.com/peterhinch/micropython-nano-gui/tree/master/gui/fonts) in each of these locations
1. In the filesystem (using standard firmware).
2. In romfs (using a romfs build)
3. As frozen bytecode.

In each case run the following script:
```py
import gc
gc.collect()
mf = gc.mem_free()
import arial10
import arial35
import arial_50
import font10
import font6
import freesans20
gc.collect()
print(mf -gc.mem_free())  # 28784 bytes
```

### Expected behaviour

Expected RAM usage of about 28K when located in the filesystem, at least an order of magnitude less in the two other cases.

### Observed behaviour

Ram used - as expected apart from case 2:
1. 28,800 bytes.
2. 28,784 bytes.
3. 2,336 bytes.

The romfs result is substantially the same as in the filesystem, implying that the bytecode is executed from RAM.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

dpgeorge, 2025-04-18:
You'll need to put .mpy files in the ROMFS, rather than .py files.

Make sure you have `pip install mpy-cross` installed in your virtual env, then it'll automatically compile .py to .mpy during the deploy.

Testing on PYBD-SF2, *without* mpy-cross:
```
$ mpremote romfs deploy fonts
Building romfs filesystem, source directory: fonts/
/
|-- arial10.py
|-- arial35.py
|-- arial_50.py
|-- courier20.py
|-- font10.py
|-- font6.py
\-- freesans20.py
Warning: `mpy_cross` module not found, .py files were not precompiled
Image size is 125508 bytes
ROMFS0 partition has size 1048576 bytes
Preparing ROMFS0 partition for writing
Writing at offset 122880
ROMFS image deployed
```
Note the warning, and the total image size.

Then with mpy-cross installed via pip:
```
$ mpremote romfs deploy fonts
Building romfs filesystem, source directory: fonts/
/
|-- arial10.py -> .mpy
|-- arial35.py -> .mpy
|-- arial_50.py -> .mpy
|-- courier20.py -> .mpy
|-- font10.py -> .mpy
|-- font6.py -> .mpy
\-- freesans20.py -> .mpy
Image size is 31010 bytes
ROMFS0 partition has size 1048576 bytes
Preparing ROMFS0 partition for writing
Writing at offset 28672
ROMFS image deployed
```
Note the image size is much less, because everything is compiled.

Then running the test to see how much memory is taken up, I get 3104 bytes.

(BTW, I have been using your font-to-py converter for large fonts with ROMFS and I have had very good results, with almost 1MiB of font data executing from the ROMFS.  It wouldn't have been possible to use such large fonts without ROMFS.)

peterhinch, 2025-04-19:
> Warning: `mpy_cross` module not found, .py files were not precompiled

Thank you - I didn't spot that message. Now fixed.

closed 2025-04-19 by peterhinch