micropython/micropython #17167: build error: implicit conversion from 'float' to 'double'

permaBox, 2025-04-22:
### Port, board and/or hardware

nrf port, nrf52 board

### MicroPython version

mpy latest, nrf52 board

The build process has failed with a compiler configuration issue if support for double is configured.

Build error report:

CC ../../py/formatfloat.c
../../py/formatfloat.c: In function 'mp_format_float':
../../py/formatfloat.c:92:26: error: implicit conversion from 'float' to 'double' to match other operand of binary expression [-Werror=double-promotion]
   92 | #define fp_isless1(x) (x < 1.0)
      |                          ^
../../py/formatfloat.c:183:16: note: in expansion of macro 'fp_isless1'
  183 |     } else if (fp_isless1(f)) {
      |                ^~~~~~~~~~
../../py/formatfloat.c:336:15: error: implicit conversion from 'float' to 'double' to match other operand of binary expression [-Werror=double-promotion]
  336 |             f *= FPCONST(10.0);
      |               ^~
cc1: all warnings being treated as errors

### Reproduction

mpconfigport.h

#if NRF51
#define MICROPY_FLOAT_IMPL          (MICROPY_FLOAT_IMPL_NONE)
#else
#define MICROPY_FLOAT_IMPL          (MICROPY_FLOAT**_IMPL_DOUBLE**)
#endif

### Observed behaviour

Related build error report.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

jonnor, 2025-05-31:
Hi and thank you for reporting. Do you know if this affects other (ARM) ports as well? Can you provide a patch?

agatti, 2025-06-04:
> Hi and thank you for reporting. Do you know if this affects other (ARM) ports as well? Can you provide a patch?

That's GCC version dependent, see https://github.com/micropython/micropython/issues/16803#issuecomment-2911981944 for the same issue on x64.

agatti, 2025-06-04:
https://github.com/agatti/micropython/commit/7116f02d5f5dddfaab9b54d8d56c0f17a491f6c2 is a better fix, working for both Makefile and CMake based ports.

agatti, 2025-06-09:
@permaBox do the changes in https://github.com/agatti/micropython/commit/7116f02d5f5dddfaab9b54d8d56c0f17a491f6c2 work for you?

Also, can you please paste the output of `arm-none-eabi-gcc -v` so I can make the fix kick in for a narrower GCC versions range rather than anything before GCC15?