micropython/micropython #17170: 在构建 MicroPython for ESP32 时遇到链接错误：未定义的 esp_panic_handler_reconfigure_wdts 符号

read-April, 2025-04-23:
### Port, board and/or hardware

- MicroPython Port: esp32 - Board: ESP32S3 ) - Hardware: 	ESP32-S3-WROOM-1-N16R8

### MicroPython version

v1.25.0-24-g0b7296233

### Reproduction

1. 使用以下命令克隆 MicroPython 仓库：
   git clone https://github.com/micropython/micropython.git
   cd micropython/ports/esp32

2. 设置 ESP-IDF 环境（release/v5.4.1）：
 cd esp-idf
./install.sh esp32 
source export.sh   

3. 构建项目：
$ make -C mpy-cross
$ cd ports/esp32
$ make submodules
$ make
$ make BOARD=ESP32_GENERIC_S3

### Expected behaviour

预期构建过程顺利完成，生成可用于 ESP32 的 MicroPython 固件。

### Observed behaviour

在链接阶段出现以下错误：
/home/read/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(panichandler.c.obj): in function `__wrap_esp_panic_handler':
/home/read/esp32/micropython/ports/esp32/build-ESP32_GENERIC_S3/../panichandler.c:49:(.text.__wrap_esp_panic_handler+0x6): undefined reference to `esp_panic_handler_reconfigure_wdts'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

### Additional Information

No, I've provided everything above.- 经过调查，发现 `esp_panic_handler_reconfigure_wdts` 在 ESP-IDF 的 commit cd887ef59a7b966a7f431754aaec6ee653849d77 中被移除。

### Code of Conduct

Yes, I agree

Josverl, 2025-04-23:
likely related : 
- https://github.com/espressif/esp-idf/issues/15687
- https://github.com/micropython/micropython/pull/17055

read-April, 2025-04-24:
Thank you for your suggestion! I understand that using English can facilitate better communication and help in getting support. Below is a brief description of the issue:

## Issue Description:
During the build process of MicroPython firmware, the following error occurred at the linking stage:

```
undefined reference to `esp_panic_handler_reconfigure_wdts'
```
Upon investigation, I found that this function was removed in ESP-IDF

## Steps to Reproduce:
- Clone the MicroPython repository and navigate to the ports/esp32 directory.
- Set up the ESP-IDF environment (version release/v5.4.1).
- Run the build commands:
```
make -C mpy-cross
make submodules
make BOARD=ESP32_GENERIC_S3
```

## Expected Behavior:
The build process completes successfully, generating MicroPython firmware for ESP32.

## Observed Behavior:
The above error occurred during the linking stage.

## Additional Information:
MicroPython version: v1.25.0-24-g0b7296233

Could you guide me on how to modify the MicroPython firmware for the ESP32-S3-WROOM-1-N16R8 module? Which specific contents need to be adjusted or modified to complete this process, and what methods can be used to optimize the performance of this module

read-April, 2025-04-24:
Thank you for the detailed explanation! I understand that ESP-IDF 5.4.1 is not yet supported, so I will use one of the supported versions. For the ESP32-S3 module, I will follow your suggestion and use the SPIRAM_OCT variant for configuration.

I also noticed the default limitation of Flash to 8MiB. To expand it to 16MiB, I will refer to the relevant files in /ports/esp32/boards/ESP32_GENERIC_S3 and try to create a custom variant to accommodate a larger partition layout.

Thanks again for your help!