micropython/micropython #17180: Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

robtinkers, 2025-04-24:
### Port, board and/or hardware

esp32 port

### MicroPython version

Official 1.25 (not a regression -- had the same issue on 1.24.1)

### Reproduction

I am writing an async web-server and get a guru meditation error when using generators. Non-generator pages seem to work fine.

Unfortunately I'm not really able to edit my code down to a minimal version, but the methods in question are below.

Full code is available if that would be helpful.

```
    async def _countdown_generator(self, n):
        try:
            print("QQ 1")
            # Send initial HTML structure
            yield f'''<!doctype html>
<html>
<head><title>Countdown from {n}</title></head>
<body>
<h1>Countdown from {n}</h1>
'''
            print("QQ 2")
            await asyncio.sleep(0.1)
            print("QQ 3")

            # Countdown loop
            for i in range(n, 0, -1):
                print("QQ 4", i)
                yield f'<p>{i}</p>\n'
                print("QQ 5", i)
                await asyncio.sleep(1)
                print("QQ 6", i)

            print("QQ 7")
            yield '<p>Done!</p>\n'
            yield '<p><a href="/">Back to Home</a></p>\n'
            yield '</body></html>'
            print("QQ 8")
        except asyncio.CancelledError:
             self.log("Countdown generator cancelled.")
             raise
        except Exception as e:
             self.log(f"Error in countdown generator: {e}")
             yield f"<p>Error during countdown: {e}</p></body></html>"
        print("QQ 9")
```

and

```
        if isinstance(content, (bytes, bytearray, memoryview)):
            writer.write(content)
        elif isiterator(content):  # also includes generators and file handles from open()
            async def _write_chunk(writer, chunk):
                print('ZZ _write_chunk() called', (repr(chunk[:20]) + ' ...') if chunk and len(chunk) > 20 else repr(chunk))
                print('ZZ mem_free() =', gc.mem_free())
                print('ZZ HEAP_DATA =', repr(esp32.idf_heap_info(esp32.HEAP_DATA)))
                print('ZZ HEAP_EXEC =', repr(esp32.idf_heap_info(esp32.HEAP_EXEC)))
                if isinstance(chunk, str):
                    chunk = chunk.encode('utf-8')
                elif isinstance(chunk, (bytes, bytearray, memoryview)):
                    pass
                elif chunk is not None:
                    chunk = bytes(chunk)
                if chunk:  # zero-length chunks mark the end, and this is not the end
                    writer.write(hex(len(chunk))[2:].encode('ascii'))
                    writer.write(b'\r\n')
                    writer.write(chunk)
                    writer.write(b'\r\n')
                    await writer.drain()
                print('ZZ _write_chunk() complete')
            
            if hasattr(content, 'read'):
                while True:
                    chunk = content.read(4096)
                    if not chunk:
                        break
                    await _write_chunk(writer, chunk)
            else:
                for chunk in content:
                    await _write_chunk(writer, chunk)
            writer.write(b'0\r\n\r\n')
        await writer.drain()
```

### Observed behaviour

The browser does get as far as showing the `<p>5</p>`.

Thonny console:
```
MPY: soft reboot
Connecting to 'xxx' ..
Connected! ('192.168.0.148', '255.255.255.0', '192.168.0.1', '192.168.0.1')
[MyTestApp] TestApp initialized.
[MyTestApp] Starting server...
[MyTestApp] Server listening on 0.0.0.0:80
[MyTestApp] 1073554544 Connection from ('192.168.0.147', 61501)
[MyTestApp] Request: GET /countdown/5
[MyTestApp] Starting countdown from 5
QQ 1
ZZ _write_chunk() called '<!doctype html>\n<htm' ...
ZZ mem_free() = 116752
ZZ HEAP_DATA = [(240, 4, 0, 4), (7288, 4, 0, 4), (16648, 4, 0, 4), (85192, 4, 0, 4), (15072, 4, 0, 4), (113840, 91760, 90112, 91656)]
ZZ HEAP_EXEC = [(15072, 4, 0, 4), (113840, 91760, 90112, 91656), (14916, 14476, 14336, 14444)]
ZZ _write_chunk() complete
QQ 2
ZZ _write_chunk() called None
ZZ mem_free() = 115888
ZZ HEAP_DATA = [(240, 4, 0, 4), (7288, 4, 0, 4), (16648, 4, 0, 4), (85192, 4, 0, 4), (15072, 4, 0, 4), (113840, 91760, 90112, 91656)]
ZZ HEAP_EXEC = [(15072, 4, 0, 4), (113840, 91628, 90112, 91612), (14916, 14476, 14336, 14444)]
ZZ _write_chunk() complete
QQ 3
QQ 4 5
ZZ _write_chunk() called '<p>5</p>\n'
ZZ mem_free() = 115088
ZZ HEAP_DATA = [(240, 4, 0, 4), (7288, 4, 0, 4), (16648, 4, 0, 4), (85192, 4, 0, 4), (15072, 4, 0, 4), (113840, 91760, 90112, 91612)]
ZZ HEAP_EXEC = [(15072, 4, 0, 4), (113840, 91744, 90112, 91612), (14916, 14476, 14336, 14444)]
ZZ _write_chunk() complete
QQ 5 5
ZZ _write_chunk() called None
ZZ mem_free() = 114240
ZZ HEAP_DATA = [(240, 4, 0, 4), (7288, 4, 0, 4), (16648, 4, 0, 4), (85192, 4, 0, 4), (15072, 4, 0, 4), (113840, 91636, 90112, 91612)]
ZZ HEAP_EXEC = [(15072, 4, 0, 4), (113840, 93328, 90112, 91472), (14916, 14476, 14336, 14444)]
ZZ _write_chunk() complete
QQ 6 5
QQ 4 4
ZZ _write_chunk() called '<p>4</p>\n'
ZZ mem_free() = 113440
ZZ HEAP_DATA = [(240, 4, 0, 4), (7288, 4, 0, 4), (16648, 4, 0, 4), (85192, 4, 0, 4), (15072, 4, 0, 4), (113840, 93328, 90112, 91472)]
ZZ HEAP_EXEC = [(15072, 4, 0, 4), (113840, 93328, 90112, 91472), (14916, 14476, 14336, 14444)]

A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

MPY version : v1.25.0 on 2025-04-15
IDF version : v5.2.2
Machine     : Generic ESP32 module with ESP32

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4012d2f2  PS      : 0x00060930  A0      : 0x8012d397  A1      : 0x3ffcf000  
A2      : 0x40104040  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x3ffd4f40  A7      : 0x3ffd4f40  A8      : 0x8012d316  A9      : 0x00000000  
A10     : 0x3ffd4f40  A11     : 0x3ffd4f40  A12     : 0x3ffd4f40  A13     : 0x3ffd6658  
A14     : 0x3ffd6290  A15     : 0x3ffbc984  SAR     : 0x00000006  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000c  LBEG    : 0x40082f15  LEND    : 0x40082f1d  LCOUNT  : 0x00000027  

Backtrace: 0x4012d2ef:0x3ffcf000 0x4012d394:0x3ffcf030 0x401041b1:0x3ffcf050 0x400f7751:0x3ffcf070 0x400fed71:0x3ffcf090 0x400fee39:0x3ffcf0b0 0x40084d4e:0x3ffcf0d0 0x400f7877:0x3ffcf170 0x400fed71:0x3ffcf1a0 0x40084cab:0x3ffcf1c0 0x400f7877:0x3ffcf260 0x400fed71:0x3ffcf2c0 0x400fee39:0x3ffcf2e0 0x40084d4e:0x3ffcf300 0x400f7877:0x3ffcf3a0 0x400fed71:0x3ffcf3d0 0x400fee39:0x3ffcf3f0 0x40084d4e:0x3ffcf410 0x400f7877:0x3ffcf4b0 0x400fed71:0x3ffcf4e0 0x400fed86:0x3ffcf500 0x4010cdaf:0x3ffcf520 0x4010cef2:0x3ffcf5b0 0x400f052e:0x3ffcf600

ELF file SHA256: b93509347

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4892
ho 0 tail 12 room 4
load:0x40078000,len:14896
load:0x40080400,len:4
load:0x40080404,len:3372
entry 0x400805b0
MicroPython v1.25.0 on 2025-04-15; Generic ESP32 module with ESP32
Type "help()" for more information.
```

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

agatti, 2025-05-08:
Out of curiosity, does the issue go away if you build your own version of MicroPython with [these code modifications](https://github.com/micropython/micropython/issues/12871#issuecomment-2389057032) to the asyncio module?

robtinkers, 2025-05-13:
Thanks for taking a look. After some more research, I think my core problem might be trying to use a language feature (async generators) that doesn't exist in micropython.

There is some old code to implement it https://github.com/micropython/micropython/pull/6668 which I thought I had managed to update to v1.25.0, but the build fails at the final `make` with

```make[3]: *** [esp-idf/main/CMakeFiles/BUILD_FROZEN_CONTENT.dir/build.make:72: esp-idf/main/CMakeFiles/BUILD_FROZEN_CONTENT] Error 1
make[2]: *** [CMakeFiles/Makefile2:5793: esp-idf/main/CMakeFiles/BUILD_FROZEN_CONTENT.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:136: all] Error 2```

which I wouldn't know how to fix.

But as you say, correct behaviour is obviously not a crash.

I did get actual minimal code to reproduce the bug, but then mislaid that test board. I'll have another dig around at home.