micropython/micropython #17181: esp32-c3 espnow restart!

NEChina, 2025-04-24:
### Port, board and/or hardware

esp32-c3

### MicroPython version

MicroPython v1.25.0

### Reproduction

When using the official firmware versions 1.24.1 to 1.25.0 on the ESP32-C3, the device automatically restarts when the ESPNOW module is used.

firm
ware: https://micropython.org/resources/firmware/ESP32_GENERIC_C3-20250415-v1.25.0.bin
code:
import network
import espnow

sta = network.WLAN(network.WLAN.IF_STA)
sta.active(True)
sta.disconnect()
e = espnow.ESPNow()
e.active(True)
peer = b'\x9c\x9e\x6e\x31\x35\x6c'
e.add_peer(peer) 
e.send(peer, "Starting...")    <---restart!

### Observed behaviour

MicroPython v1.25.0 on 2025-04-19; ESP32C3 module with ESP32C3

Type "help()" for more information.

>>> import network
>>> import espnow
>>> sta = network.WLAN(network.WLAN.IF_STA)
>>> sta.active(True)
True
>>> sta.disconnect()
>>> e = espnow.ESPNow()
>>> e.active(True)
True
>>> peer = b'\x9c\x9e\x6e\x31\x35\x6c'
>>> e.add_peer(peer) 
>>> e.send(peer, "Starting...")
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048b82
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0xd9c
load:0x403cc710,len:0x9a8
load:0x403ce710,len:0x2c3c
PROBLEM IN THONNY'S BACK-END: Exception while handling 'execute_source' (ConnectionError: EOF).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

agatti, 2025-04-25:
> MicroPython v1.25.0 on 2025-04-19; ESP32C3 module with ESP32C3

Is that a typo?  It should read `2025-04-15`...

Just to be sure, did you perform a full flash erase cycle before flashing the image?  This is how boot messages like when booting on my board:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0xf28
load:0x403cc710,len:0x944
load:0x403ce710,len:0x2b1c
entry 0x403cc710
MicroPython v1.25.0 on 2025-04-15; ESP32C3 module with ESP32C3
Type "help()" for more information.
```

Note how the load segments have different lengths, too.

If even with a full erase/reflash cycle things still break, please download the file again and retry (that file should be 1835856 bytes long and its MD5 is e61f4aa8e8ca0950482f83185fa597d8, along with its SHA256 being 4dd82c4bc8b90d4a8f303eedc1bf2b87ace1827daad3a855c6ce3209482f46f3), maybe you've uploaded a partial firmware file due to a failed transfer.

NEChina, 2025-04-25:
Sorry, I used my own compiled version before. I just flashed the official version, the same test code, and it also restarted.
MicroPython v1.25.0 on 2025-04-15; ESP32C3 module with ESP32C3

Type "help()" for more information.

>>> %run -c $EDITOR_CONTENT
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038066c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0xf28
load:0x403cc710,len:0x944
load:0x403ce710,len:0x2b1c
entry 0x403cc710
PROBLEM IN THONNY'S BACK-END: Exception while handling 'run' (ConnectionError: EOF).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.

Process ended with exit code 1.

NEChina, 2025-04-25:
I also find it very strange that not only the espnow test code causes the reboot, but also the wifi ap code. Is it possible that the wifi function of my module is broken?
I bought two esp32-c3s, and both of them have the same problem.

code:
import network
import espnow

# sta = network.WLAN(network.WLAN.IF_STA)
# sta.active(True)
# e = espnow.ESPNow()
# e.active(True)
# peer = b'\x9c\x9e\x6e\x31\x35\x6c'
# e.add_peer(peer) 
# 
# e.send(peer, "Starting...")    <---restart!

ap = network.WLAN(network.WLAN.IF_AP) 
ap.config(ssid='ESP-AP')              
ap.config(max_clients=10)             
ap.active(True)                     <---restart!

agatti, 2025-04-25:
Well, now things are clearer for sure :)  If you dig a little into the small output fragment you pasted you'll see that the last program counter mentioned is inside `esp-idf/esp_system/libesp_system.a(brownout.c.obj)`, which strongly points to a power issue on your board.

This looks like it's the same issue as #16826 - can you try the same workarounds (as in, lower wifi transmission power) and see if that makes the issue go away?

NEChina, 2025-04-25:
> Well, now things are clearer for sure :) If you dig a little into the small output fragment you pasted you'll see that the last program counter mentioned is inside [...]

Thanks to **agatti** for his help.
It turned out that the restart was caused by the power supply problem. When I designed the circuit, I used RS3236-3.3 5v-3.3v Max500mA, and it was only used for the module, and no other sensors used this DC-DC. I don't understand why this would cause problems.

Solution:
import network
import espnow

```
sta = network.WLAN(network.WLAN.IF_STA)
sta.active(True)
sta.config(txpower=15)               <--add
e = espnow.ESPNow()
e.active(True)
peer = b'\x9c\x9e\x6e\x31\x35\x6c'
e.add_peer(peer) 

e.send(peer, "Starting...")
#e.send(peer, b'end')

# ap = network.WLAN(network.WLAN.IF_AP) 
# ap.config(ssid='ESP-AP')                     
# ap.active(True)         
# ap.config(txpower=15)            <--add
```

NEChina, 2025-04-25:
Thanks to **agatti** for help.

closed 2025-04-25 by NEChina