micropython/micropython #17198: Bootloop when using 64MBit flash with ESP8266

mikhan253, 2025-04-27:
### Port, board and/or hardware

ESP8266 ESP-07 with W25Q64F

### MicroPython version

ESP8266_GENERIC-FLASH_2M_ROMFS-20250415-v1.25.0.bin
ESP8266_GENERIC-FLASH_1M-20250415-v1.25.0.bin
ESP8266_GENERIC-OTA-20250415-v1.25.0.bin
ESP8266_GENERIC-20250415-v1.25.0.bin

### Reproduction

programmed with the following commands:

esptool --port COM3 --baud 115200 write_flash --erase-all --flash_size=detect -fm dout 0 ESP8266_GENERIC-FLASH_2M_ROMFS-20250415-v1.25.0.bin
esptool.py v4.8.1
Serial port COM3
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 18:fe:34:9d:5c:bb
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 8MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 16.0s
Flash params set to 0x0380
Compressed 639680 bytes to 428052...
Wrote 639680 bytes (428052 compressed) at 0x00000000 in 37.8 seconds (effective 135.5 kbit/s)...
Hash of data verified.

### Observed behaviour

Initially starts with the message: "Writing init data" and then bootloops forever.

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 31264, room 16
tail 0
chksum 0x7a
load 0x3ffe8000, len 1060, room 8
tail 12
chksum 0xc8
ho 0 tail 12 room 4
load 0x3ffe8430, len 1124, room 12
tail 8
chksum 0x2b
csum 0x2b
Writing init data
rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 31264, room 16
tail 0
chksum 0x7a
load 0x3ffe8000, len 1060, room 8
tail 12
chksum 0xc8
ho 0 tail 12 room 4
load 0x3ffe8430, len 1124, room 12
tail 8
chksum 0x2b
csum 0x2b
rf_cal[0] !=0x05,is 0xFF
....
....

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

mikhan253, 2025-04-27:
Thank you Robert, i have tried now like the following:

esptool --port COM3 --after no_reset --baud 460800 write_flash --erase-all 0x0 ESP8266_GENERIC-FLASH_2M_ROMFS-20250415-v1.25.0.bin 0x7fc000 esp_init_data_default.bin
esptool.py v4.8.1
Serial port COM3
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 18:fe:34:9d:5c:bb
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Erasing flash (this may take a while)...
Chip erase completed successfully in 15.0s
Compressed 639680 bytes to 428052...
Wrote 639680 bytes (428052 compressed) at 0x00000000 in 9.8 seconds (effective 523.9 kbit/s)...
Hash of data verified.
Compressed 128 bytes to 72...
Wrote 128 bytes (72 compressed) at 0x007fc000 in 0.1 seconds (effective 18.9 kbit/s)...
Hash of data verified.

but it also makes the same output:
M:\Projekte\Photovoltaik\BKW_REG_V01\Micropython\Flash

esptool --port COM3 --after no_reset --baud 460800 write_flash --erase-all 0x0 ESP8266_GENERIC-FLASH_2M_ROMFS-20250415-v1.25.0.bin 0x7fc000 esp_init_data_default.bin

 ets Jan  8 2013,rst cause:2, boot mode:(3,2)

load 0x40100000, len 31264, room 16
tail 0
chksum 0x7a
load 0x3ffe8000, len 1060, room 8
tail 12
chksum 0xc8
ho 0 tail 12 room 4
load 0x3ffe8430, len 1124, room 12
tail 8
chksum 0x2b
csum 0x2b
Writing init data
rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:2, boot mode:(3,2)

load 0x40100000, len 31264, room 16
tail 0
chksum 0x7a
load 0x3ffe8000, len 1060, room 8
tail 12
chksum 0xc8
ho 0 tail 12 room 4
load 0x3ffe8430, len 1124, room 12
tail 8
chksum 0x2b
csum 0x2b
rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:2, boot mode:(3,2)

load 0x40100000, len 31264, room 16
tail 0
chksum 0x7a
load 0x3ffe8000, len 1060, room 8
tail 12
chksum 0xc8
ho 0 tail 12 room 4
load 0x3ffe8430, len 1124, room 12
tail 8
chksum 0x2b
csum 0x2b
rf_cal[0] !=0x05,is 0xFF

i even tried to write it to 0x3fc000, because maybe it thinks its a 4MB flash. But it didnt work either

robert-hh, 2025-04-27:
That looks like the "normal" boot loop error. Now try the other -fm options. 
`esptool --port COM3 --after no_reset --baud 460800 write_flash  -fm dout --flash_size=8M 0x0 ESP8266_GENERIC-FLASH_2M_ROMFS-20250415-v1.25.0.bin 0x7fc000 esp_init_data_default.bin`
The flash mode options are qio,qout,dio,dout, with the default being dio.
You do not have to erase the flash all the time.

mikhan253, 2025-04-27:
ok i tried it again with different options: 

esptool --port COM3 --after no_reset --baud 460800 write_flash --erase-all --flash_freq 80m --flash_mode qio --flash_size 8MB --verify 0x0 ESP8266_GENERIC-FLASH_2M_ROMFS-20250415-v1.25.0.bin 0x7fc000 esp_init_data_default.bin

now it works! thank you very much!!!

closed 2025-04-27 by mikhan253