micropython/micropython #17203: creating virtual timer on MicroPython v1.25.0 on esp32 s3 fails

NilsBe, 2025-04-28:
### Port, board and/or hardware

esp32 s3

### MicroPython version

MicroPython v1.25.0 on 2025-04-15; Generic ESP32S3 module with ESP32S3

### Reproduction

>>> import machine
>>> t = machine.Timer(-1)

### Expected behaviour

no error on MicroPython v1.24.1 on 2024-11-29; Generic ESP32S3 module with ESP32S3

### Observed behaviour

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: invalid Timer number

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

robert-hh, 2025-04-28:
As far as I can tell from going back the versions, there was not Softtimer support in V1.24.1 for ESP32. The supported times is and was a hard timer. Only the argument value check was improved in V1.25 to only accept values between 0 and <number_of_timers - 1>.  For the ESP32S3 its  0-3. 
With 1.24.1 and below, a timer number of -1 would result in using timer 3.
See the definitions from the esp-idf below:
```
~/Downloads/esp-idf/components/soc/esp32/include/soc/soc_caps.h:
  316: #define SOC_TIMER_GROUP_TOTAL_TIMERS      (4)
~/Downloads/esp-idf/components/soc/esp32c2/include/soc/soc_caps.h:
  266: #define SOC_TIMER_GROUP_TOTAL_TIMERS      (1U)
~/Downloads/esp-idf/components/soc/esp32c3/include/soc/soc_caps.h:
  352: #define SOC_TIMER_GROUP_TOTAL_TIMERS      (2)
~/Downloads/esp-idf/components/soc/esp32c6/include/soc/soc_caps.h:
  429: #define SOC_TIMER_GROUP_TOTAL_TIMERS      (2)
~/Downloads/esp-idf/components/soc/esp32h2/include/soc/soc_caps.h:
  424: #define SOC_TIMER_GROUP_TOTAL_TIMERS      (2)
~/Downloads/esp-idf/components/soc/esp32p4/include/soc/soc_caps.h:
  458: #define SOC_TIMER_GROUP_TOTAL_TIMERS      4
~/Downloads/esp-idf/components/soc/esp32s2/include/soc/soc_caps.h:
  323: #define SOC_TIMER_GROUP_TOTAL_TIMERS      (4)
~/Downloads/esp-idf/components/soc/esp32s3/include/soc/soc_caps.h:
  359: #define SOC_TIMER_GROUP_TOTAL_TIMERS      (4)
```

NilsBe, 2025-04-28:
I always thought I would use a virtual timer with -1.  :-) 
Thanks for your answer. 
I am now trying to figure out where I could have learned that there is no virtual timer on esp32 s3 (or is it on all esp32?) 

Kind regards

closed 2025-04-28 by NilsBe