micropython/micropython #17204: lightsleep(n) returns immediately on pico-w or pico with network imported

phdussud, 2025-04-28:
### Port, board and/or hardware

pico and pico-w

### MicroPython version

MicroPython v1.25.0 on 2025-04-15; Raspberry Pi Pico W with RP2040

### Reproduction

copy paste this code:
```
# light_sleep_test.py
from machine import lightsleep
from time import sleep_ms, ticks_ms, ticks_diff

def test(n):
    for i in range(n):
        t0 = ticks_ms()
        sleep_ms(50)
        lightsleep(5000)
        sleep_ms(50)
        print(i, ticks_diff(ticks_ms(), t0))

test(10)
```
If the test works it should print 
0 5101
1 5101
2 5101
3 5101
4 5101
5 5101
6 5101
7 5100
8 5101
9 5100

Instead it prints:
0 142
1 127
2 127
3 128
4 129
5 127
6 128
7 128
8 128
9 128

### Expected behaviour

If the test works it should print 
0 5101
1 5101
2 5101
3 5101
4 5101
5 5101
6 5101
7 5100
8 5101
9 5100

### Observed behaviour

Instead it prints:
0 142
1 127
2 127
3 128
4 129
5 127
6 128
7 128
8 128
9 128

### Additional Information

See [discussion here](https://github.com/orgs/micropython/discussions/17185#discussioncomment-12951488)

### Code of Conduct

Yes, I agree

cpottle9, 2025-04-30:
I am pretty sure this is the same issue as [16181](https://github.com/micropython/micropython/issues/16181).

I did a quick test in 1.24.1 and I see the problem their too.

phdussud, 2025-04-30:
I think this is correct. The proposed solution outlined in 16181 makes sense to me.