micropython/micropython #17217: Request: Port for Lolin ESP32-S3 Pro (16MB flash, 8MB SPRAM)

davidmcnabnz, 2025-04-30:
### Description

I notice a few ports for the ESP32 family, including a couple of lower-spec Lolin boards, plus generic ESP32-S3 ports. But these don't accommodate boards like the Lolin ESP32-S3 Pro with its 16MB Flash and 8MB PSRAM.

Is there any possibility of ports in the near future, for this board, and other newer/higher models of ESP32?

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

dobodu, 2025-05-08:
I agree,

In the meanwhile you can add an

mpconfigvariant_FLASH_16M_SPIRAM_OCT.cmake
`set(IDF_TARGET esp32s3)

set(SDKCONFIG_DEFAULTS
     boards/sdkconfig.base
     ${SDKCONFIG_IDF_VERSION_SPECIFIC}
     boards/sdkconfig.usb
     boards/sdkconfig.ble
     boards/sdkconfig.spiram_sx
     boards/ESP32_GENERIC_S3/sdkconfig.board
 )

if(MICROPY_BOARD_VARIANT STREQUAL "SPIRAM_OCT")
     set(SDKCONFIG_DEFAULTS
         ${SDKCONFIG_DEFAULTS}
         boards/sdkconfig.240mhz
         boards/sdkconfig.spiram_oct
     )

     list(APPEND MICROPY_DEF_BOARD
         MICROPY_HW_BOARD_NAME="Generic ESP32S3 module with Octal-SPIRAM"
     )
endif()

if(MICROPY_BOARD_VARIANT STREQUAL "FLASH_4M")
     set(SDKCONFIG_DEFAULTS
         ${SDKCONFIG_DEFAULTS}
         boards/ESP32_GENERIC_S3/sdkconfig.flash_4m
     )
endif()

if(MICROPY_BOARD_VARIANT STREQUAL "FLASH_16M_SPIRAM_OCT")
     set(SDKCONFIG_DEFAULTS
         ${SDKCONFIG_DEFAULTS}
         boards/sdkconfig.240mhz
         boards/sdkconfig.spiram_oct
         boards/ESP32_GENERIC_S3/sdkconfig.flash_16m
     )

     list(APPEND MICROPY_DEF_BOARD
         MICROPY_HW_BOARD_NAME="ESP32S3 microcontroller with 16MB of Flash & Octal-SPIRAM"
     )
endif()`

along with 

sdkconfig.flash_16m
`                                                             
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=
CONFIG_ESPTOOLPY_FLASHSIZE_8MB=
CONFIG_ESPTOOLPY_FLASHSIZE_16MB=y

CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions-16MiB.csv"
`

and compile with

make  BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=FLASH_16M_SPIRAM_OCT

mattytrentini, 2025-05-13:
I've submitted PR #17290 to define a board for the Lolin S3 Pro; please test/comment if you can!