micropython/micropython #17219: gzip compression with requests.get() not supported?

juhanijaakola, 2025-04-30:
### Port, board and/or hardware

ESP32_GENERIC_C3-20250415-v1.25.0

### MicroPython version

MicroPython v1.25.0 on 2025-04-15; ESP32C3 module with ESP32C3

### Reproduction

import requests
url = 'https://tie.digitraffic.fi/api/weather/v1/stations/2020/data' 
headers = {'Digitraffic-User': 'NOT DISCLOSED'}
response = requests.get(url,headers=headers)

### Expected behaviour

In a PC running Linux (Fedora 40) and Python 3.13.2:

**response = requests.get(url,headers=headers)**
**response.status_code**
200
**response.reason**
'OK'
**response.headers**
{'Content-Type': 'application/json;charset=UTF-8',
'Transfer-Encoding': 'chunked', 'Connection': 'keep-alive', 'Date':
'Tue, 15 Apr 2025 09:23:06 GMT', 'Server': 'openresty',
'Last-Modified': 'Tue, 15 Apr 2025 09:20:10 GMT', 'ETag':
'W/"09f42986e01c2306e4bebc097160cff60"', 'Expires': 'Tue, 15 Apr 2025
09:24:06 GMT', 'Cache-Control': 'max-age=60',
'access-control-allow-origin': '*', 'access-control-allow-methods':
'GET, POST, OPTIONS', 'access-control-allow-headers':
'DNT,X-CustomHeader,Keep-Alive,User-Agent,X-Requested-With,If-Modified-Since,Cache-Control,Content-Type,Content-Range,Range,Digitraffic-User',
'access-control-expose-headers':
'DNT,X-CustomHeader,Keep-Alive,User-Agent,X-Requested-With,If-Modified-Since,Cache-Control,Content-Type,Content-Range,Range,Digitraffic-User',
**'Content-Encoding': 'gzip'**, 'X-Cache': 'Miss from cloudfront', 'Via':
'1.1 [14b5d848e0a4cab1de054891ea1e787c.cloudfront.net](http://14b5d848e0a4cab1de054891ea1e787c.cloudfront.net/) (CloudFront)',
'X-Amz-Cf-Pop': 'HEL51-P1', 'Alt-Svc': 'h3=":443"; ma=86400',
'X-Amz-Cf-Id': '67LEquPoJdsZmDbo3Uyjg7BjMmfh6RqcG9I36LhifjvONyR3W0fC7g=='}

### Observed behaviour

In MicroPython 1.24.1 and 1.25.0 on a ESP32-C3:

**response = requests.get(url,headers=headers)**
**response.status_code**
406
**response.reason**
b'Not Acceptable'
**response.headers**
{'Vary': 'Origin', 'Connection': 'close', 'Content-Length': '70',
'Via': '1.1 [1be5216f770ec05deb91e9e25b61b898.cloudfront.net](http://1be5216f770ec05deb91e9e25b61b898.cloudfront.net/)
(CloudFront)', 'X-Cache': 'Miss from cloudfront', 'X-Amz-Cf-Pop':
'HEL51-P1', 'Server': 'CloudFront', 'Alt-Svc': 'h3=":443"; ma=86400',
'X-Amz-Cf-Id': 'AVTra_4NUASUqg3TDQcLxIc9np2vn7V1fGqCEXRUTomDTefEdurI_g==',
'Date': 'Tue, 15 Apr 2025 09:20:33 GMT'}
**response.text**
'Use of gzip compression is required with Accept-Encoding: gzip header.'

I have tried the following header values in my request but the result
is always the same:

'Accept-Encoding': 'gzip'
'Accept-Encoding': 'gzip, deflate'
'Accept-Encoding': 'identity'
'Accept-Encoding': 'deflate'
'Accept-Encoding': '*'

### Additional Information

The server that I'm calling seems to require using gzip. And it works in Python but not in MicroPython.

### Code of Conduct

Yes, I agree

felixdoerre, 2025-05-13:
This server seems to require HTTP 1.1, and if a client requests with the protocol version http 1.0, it responds with that message regardless of the provided `Accept-Encoding`-header: https://github.com/micropython/micropython-lib/issues/844. You can patch urequests locally to send http/1.1 as version, and then decompress using the corresponding lib manually.

jonnor, 2025-05-31:
There seems to be a proposed resolution above.

closed 2025-06-10 by dpgeorge