micropython/micropython #17224: cpython difference parsing "invalid" literals

jepler, 2025-04-30:
### Port, board and/or hardware

unix

### MicroPython version

MicroPython v1.26.0-preview.24.gcd8fad919f.dirty on 2025-04-22; linux [GCC 12.2.0] version

### Reproduction

1. `micropython -c 'print(0x1or 2)`
2. `python3.9 -c 'print(0x1or 2)`

### Expected behaviour

Both implementations print the same value

### Observed behaviour

micropython raises an exception, while python prints "1" (Python versions starting with 3.11 print a warning though)

### Additional Information

The warning about this syntax appeared in Python 3.11 as a result of https://bugs.python.org/issue43833.  There appears to be no intent to actually convert this to a SyntaxError, though only for the risk of breaking existing code, not because anyone likes the construct.

Sorry for not filing this bug on April 1. Some post on mastodon alerted me to the fact that `0x1for....real` is valid Python and I just had to check it out on micropython.

### Code of Conduct

Yes, I agree

projectgus, 2025-05-08:
Wow, interesting! I was unaware of this.

This feels like something to add to the `tests/cpydiff` so it's documented as such, rather than to fix. If we've gotten this far without anyone relying on it then it's probably better to continue not supporting it!

jepler, 2025-05-09:
actually this is already in cpydiff 🥺 though the different but related `0x1.to_bytes()` was not covered; also `1_` (trailing underscore) is rejected as a number literal in CPython but accepted in MicroPython.

jepler, 2025-05-10:
I'll close this up since it was already in cpydiff, but #17277 offers some other improvements in that direction.

closed 2025-05-10 by jepler