micropython/micropython #17229: RP2350 Watchdog and lightsleep issue

cpottle9, 2025-04-30:
### Port, board and/or hardware

Raspberry Pi Pico2

### MicroPython version

MicroPython v1.25.0 on 2025-04-15; Raspberry Pi Pico2 with RP2350

### Reproduction

Copy the following into wdt_lightsleep_2350.py.
```py
from machine import lightsleep, WDT
from time import sleep_ms

wdt = WDT(timeout=5000)

lightsleep(3000)

sleep_ms(3000)
print('Should not print')
sleep_ms(3000)
```
Do 'mpremote run wdt_lightsleep_2350.py'.

### Expected behaviour

Expect the same behavior seen when the lightsleep is replaced with sleep_ms.
File wdt_sleep_ms_2350.py
```py
from machine import lightsleep, WDT
from time import sleep_ms

wdt = WDT(timeout=5000)

sleep_ms(3000)

sleep_ms(3000)
print('Should not print')
sleep_ms(3000)
```
```
>> mpremote run wdt_sleep_ms_2350.py
Traceback (most recent call last):
  File "/home/picompute/.local/bin/mpremote", line 8, in <module>
    sys.exit(main())
```
Note: most of traceback removed.
This time the print did not execute because the watchdog timer triggered during the second sleep_ms(3000).

### Observed behaviour

The print statement does execute and the watchdog timer triggers during the last sleep_ms(3000).

```
>> mpremote run wdt_lightsleep_2350.py
Should not print
Traceback (most recent call last):
  File "/home/picompute/.local/bin/mpremote", line 8, in <module>
    sys.exit(main())
```
Note: most of traceback removed.

### Additional Information

I have changes to lightsleep in modmachine.c which fix this problem.
I will prepare a pull request with this fix.

The root cause is the watchdog timer does not count down while lightsleep() is running. 

### Code of Conduct

Yes, I agree

alke, 2025-05-04:
in RP2350 lightsleep is not working it just pass with no sleep

from machine import lightsleep
from time import sleep_ms
led = machine.Pin("LED", machine.Pin.OUT)
while True:
    led.on()
    lightsleep(3000)
#    sleep_ms(3000)
    led.off()
    sleep_ms(3000)

alke, 2025-05-04:
sorry for the hashtag this is the code with no remark
from machine import lightsleep
from time import sleep_ms
led = machine.Pin("LED", machine.Pin.OUT)
while True:
  led.on()
  lightsleep(3000)
  led.off()
  sleep_ms(3000)

cpottle9, 2025-05-04:
@alke lightsleep works fine on a Pico 2. 
The problem you mention is on the Pico 2 W (wireless) only.

I just ran your code on a Pico 2 with micropython 1.25.0 and it worked.

alke, 2025-05-05:
Sorry, yes you are right it is not working on Pico 2 W
:)