micropython/micropython #17240: nRF52840: Test extmod/machine_disable_irq.py hangs forever

Plaque-fcc, 2025-05-02:
### Port, board and/or hardware

nrf port, PROMICRO board

### MicroPython version

MicroPython v1.26.0-preview.59.g79abdad9e; PROMICRO with NRF52840

### Reproduction

1. Have **nrf** port `v1.26.0-preview.59.g79abdad9e` built and flashed (board PROMICRO in my case) onto an nRF52840 chip.
2. Connect and reset the board to make sure it's running MicroPython and isn't busy.
3. cd to `tests` and run `./run-tests.py -t /dev/ttyACM<port_number>`.
4. Some tests willl run just fine (like `extmod/machine1.py`), others should either be skipped or fail.
5. In reality, this test will fail to fail and only hang.

### Expected behaviour

Expected to pass this test. But this port probably doesn't support it yet.

### Observed behaviour

$ strace ./run-tests.py -t dev/ttyACM0

```
…
openat(AT_FDCWD, "/micropython/tests/extmod/machine_disable_irq.py", O_RDONLY|O_CLOEXEC) = 8
fstat(8, {st_mode=S_IFREG|0664, st_size=663, ...}) = 0
ioctl(8, TCGETS, 0xfffff35d8b20)        = -1 ENOTTY (Inappropriate ioctl for device)
lseek(8, 0, SEEK_CUR)                   = 0
lseek(8, 0, SEEK_CUR)                   = 0
fstat(8, {st_mode=S_IFREG|0664, st_size=663, ...}) = 0
read(8, "# Test executing Python code whi"..., 664) = 663
read(8, "", 1)                          = 0
close(8)                                = 0
write(3, "\r\3", 2)                     = 2
pselect6(7, [6], [3], NULL, NULL, NULL) = 1 (out [3])
ioctl(3, FIONREAD, [1])                 = 0
pselect6(5, [3 4], NULL, NULL, NULL, NULL) = 1 (in [3])
read(3, ">", 1)                         = 1
ioctl(3, FIONREAD, [0])                 = 0
write(3, "\r\1", 2)                     = 2
pselect6(7, [6], [3], NULL, NULL, NULL) = 1 (out [3])
pselect6(5, [3 4], NULL, NULL, NULL, NULL) = 1 (in [3])
read(3, "r", 1)                         = 1
ioctl(3, FIONREAD, [26])                = 0
pselect6(5, [3 4], NULL, NULL, NULL, NULL) = 1 (in [3])
read(3, "a", 1)                         = 1
ioctl(3, FIONREAD, [25])                = 0
pselect6(5, [3 4], NULL, NULL, NULL, NULL) = 1 (in [3])
read(3, "w", 1)                         = 1
ioctl(3, FIONREAD, [24])                = 0
pselect6(5, [3 4], NULL, NULL, NULL, NULL) = 1 (in [3])
read(3, " ", 1)                         = 1
ioctl(3, FIONREAD, [23])                = 0
pselect6(5, [3 4], NULL, NULL, NULL, NULL) = 1 (in [3])
read(3, "R", 1)                         = 1
ioctl(3, FIONREAD, [22])                = 0
pselect6(5, [3 4], NULL, NULL, NULL, NULL) = 1 (in [3])
read(3, "E", 1)                         = 1
ioctl(3, FIONREAD, [21])                = 0
pselect6(5, [3 4], NULL, NULL, NULL, NULL) = 1 (in [3])
read(3, "P", 1)                         = 1
ioctl(3, FIONREAD, [20])                = 0
pselect6(5, [3 4], NULL, NULL, NULL, NULL) = 1 (in [3])
read(3, "L", 1)                         = 1
ioctl(3, FIONREAD, [19])                = 0
pselect6(5, [3 4], NULL, NULL, NULL, NULL) = 1 (in [3])
read(3, ";", 1)                         = 1
ioctl(3, FIONREAD, [18])                = 0
[... 227 lines cut ...]
pselect6(7, [6], [3], NULL, NULL, NULL) = 1 (out [3])
pselect6(5, [3 4], NULL, NULL, NULL, NULL) = 1 (in [3])
read(3, "\1", 1)                        = 1
ioctl(3, FIONREAD, [0])                 = 0
write(3, "ge(5):\n        f(nest)\n", 23) = 23
pselect6(7, [6], [3], NULL, NULL, NULL) = 1 (out [3])
write(3, "\4", 1)                       = 1
pselect6(7, [6], [3], NULL, NULL, NULL) = 1 (out [3])
pselect6(5, [3 4], NULL, NULL, NULL, NULL) = 1 (in [3])
read(3, "\1", 1)                        = 1
ioctl(3, FIONREAD, [0])                 = 0
clock_nanosleep(CLOCK_MONOTONIC, TIMER_ABSTIME, {tv_sec=52070, tv_nsec=642399969}, NULL) = 0
ioctl(3, FIONREAD, [1])                 = 0
pselect6(5, [3 4], NULL, NULL, NULL, NULL) = 1 (in [3])
read(3, "\4", 1)                        = 1
pselect6(5, [3 4], NULL, NULL, NULL, NULL                               
```

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

dpgeorge, 2025-05-02:
This should be fixed by #17213.

Plaque-fcc, 2025-05-02:
> This should be fixed by [#17213](https://github.com/micropython/micropython/pull/17213).

MicroPython v1.26.0-preview.46.g9ac710a28; PROMICRO with NRF52840
Branch **dpgeorge/nrf-use-common-machine-irq-function**.

Left only this test enabled:

$ ./run-tests.py -t /dev/ttyACM0 
> platform=nrf52840-ProMicro arch=armv7emsp inlineasm=thumb
> pass  extmod/machine_disable_irq.py 
> 1 tests performed (3 individual testcases)
> 1 tests **passed**

Thanks, everything works!!