micropython/micropython #17241: RP2350B Core Board: says GPIO pin > 31 declared mode = OUTPUT  is shown as INPUT

Flipje1955, 2025-05-02:
### Port, board and/or hardware

RP2350B Core WeAct Studio

### MicroPython version

Using MicroPython v1.25.0 on 2025-04-15; WeAct Studio RP2350B Core with RP2350:

### Reproduction

'''
pin = Pin(31, Pin.OUT)
pin.on()
print(pin, pin.value())
pin.off()
print(pin, pin.value())

pin = Pin(32, Pin.OUT)
pin.on()
print(pin, pin.value())
pin.off()
print(pin, pin.value())
'''

### Expected behaviour

Expected all pins defined as OUTPUT to be shown as such. 

### Observed behaviour

Output:

MPY: soft reboot
Pin(GPIO31, mode=OUT) 1
Pin(GPIO31, mode=OUT) 0
Pin(GPIO32, mode=IN) 1
Pin(GPIO32, mode=IN) 0

This applies to all pins #32  up to # 47. 

nb. Same behaviour on [v1.26.0-preview.59.g79abdad9e (2025-05-02) .uf2](https://micropython.org/resources/firmware/WEACTSTUDIO_RP2350B_CORE-20250502-v1.26.0-preview.59.g79abdad9e.uf2)

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

Gadgetoid, 2025-05-12:
Looks like something I overlooked when porting to the RP2350. I think this relates to the macros defined in the top of `machine_pin.c` which directly access registers:

https://github.com/micropython/micropython/blob/69993daa5c2eae6055f0b3b2330e95e78d6e3738/ports/rp2/machine_pin.c#L49-L53

Specifically `GPIO_IS_OUT` which ignores the existence of `gpio_hi_oe` .

Should be a quick fix.

dpgeorge, 2025-05-21:
Fixed by 28c8fff6d8f920eb6199cc5db59da1baf1b00843