micropython/micropython #17242: wiznet5k_send_ethernet: fatal error -5

masker, 2025-05-03:
### Port, board and/or hardware

Wiznet W5100S-EVB-Pico

### MicroPython version

MicroPython v1.25.0 on 2025-04-15; W5100S-EVB-Pico with RP2040
also 1.24.1, and 1.24.0, but 1.23.0 doesn't give the error

### Reproduction

MicroPython v1.25.0 on 2025-04-15; W5100S-EVB-Pico with RP2040

Type "help()" for more information.

>>> import network
>>> nic = network.WIZNET5K()
>>> nic.ifconfig()
('192.168.1.137', '255.255.255.0', '192.168.1.254', '192.168.1.254')
>>> nic.active(True)
wiznet5k_send_ethernet: fatal error -5
>>> nic.isconnected()
True
>>> 

### Expected behaviour

Expected no error to be reported.
However, the error does not seem to affect the operation. The ethernet socket behaviour seems to be working ok. 

### Observed behaviour

wiznet5k_send_ethernet: fatal error -5  
was printed out, but operation of ethernet seems to be working, however I can't be certain all functions are working correctly at this time

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

Lobo-T, 2025-05-20:
I can confirm that this happens in MicroPython v1.26.0-preview.120 with a W5500

But it seems very random.
I've tried to see if there is any difference from a power on reset, soft reset.  Explicitly resetting the W5500 before creating the WIZNET5K object.
Waiting between creating the object and calling Activate or not.
Nothing seems to make a difference, sometimes there's no error, somtimes there is an error.
If the error has first occured it might be there againg next time you run the script, but might dissappear on the third or fourth time.

Most often you get the `wiznet5k_send_ethernet: fatal error -5` only once when calling` active(True)`, but once it printed three times.

The interface still seems to work though.

Testscript:

```py
from microdot import Microdot
from machine import Pin,SPI
import network
import time

spiBus = 0
spiCLK = 18
spiMOSI = 19
spiMISO = 16
ethCS = 17
ethRST = 28
ethINT = 20
pRST = Pin(ethRST,Pin.OUT)
staticIP = '10.0.250.55'

spinet = SPI(spiBus, 10_000_000, sck=Pin(spiCLK), mosi=Pin(spiMOSI), miso=Pin(spiMISO))
wiznet = network.WIZNET5K(spinet,Pin(ethCS),pRST,Pin(ethINT))

print("Activate")
wiznet.active(True)
print("Set IP")
wiznet.ifconfig((staticIP, '255.255.0.0', '10.0.250.1', '8.8.8.8')) #static
#ipconf = wiznet.ifconfig("dhcp") #dhcp
print(wiznet.ifconfig())

while True:
    print('Waiting for Ethernet connection...')
    if wiznet.isconnected():
        print('Connected to Ethernet')
        print(wiznet.ifconfig())
        break
    time.sleep(2)

app = Microdot()

webpage = '''
<html>
<head>
    <title>Wiznet test</title>
    <meta http-equiv="refresh" content="2">
</head>
<body>
    <h2>Wiznet test</h2>
    <p>Reloader om 2 sek</p>
</body>
</html>
'''

@app.route('/')
async def index(request):
    return webpage, {'Content-Type':'text/html'}

app.run(port=80,debug=True)
```

Result when connecting with a webbrowser:

```
MPY: soft reboot
Activate
wiznet5k_send_ethernet: fatal error -5
Set IP
('10.0.250.55', '255.255.0.0', '10.0.250.1', '8.8.8.8')
Waiting for Ethernet connection...
Waiting for Ethernet connection...
Connected to Ethernet
('10.0.250.55', '255.255.0.0', '10.0.250.1', '8.8.8.8')
Starting async server on 0.0.0.0:80...
GET / 200
GET / 200
GET / 200
GET / 200
GET / 200
GET / 200
GET / 200
```

I can't remember running into this when the driver was first created some years ago.
But if it just works anyway, I might have been lucky and not noticed.

I think @andrewleech made the driver. Is it OK to tag him in here?

andrewleech, 2025-05-20:
I did write the driver yes, but haven't actually used it since really! I don't know what this error means really, I'm not sure if the wiznet C drivers have any more information about them

MilhouseVH, 2025-12-27:
> Has anyone else had experiences similar to mine? Were you able to find a solution? One possibility is that one of the connections between Pico IO and W5500 control pins is flaky. I have found breadboards are not the best for reliable connections! But I have not found anything that generates intermittent operation..

No, that's not been my experience, although I tend to use the W5500-EVB-PICO boards rather than the separate W5500 modules.

As mentioned, I mostly see the error this issue is dealing with after a soft reboot (while running with Thonny).

`nic.active(False)` followed by `nic.active(True)` seems to have fixed the issue for me.

Without seeing your actual code it's difficult to say what your issues might be. I would suggest starting a new Discussion thread (rather than continue to post on this Issue which may not be related/relevant) then posting on your Discussions thread a minimalist example of your code that reproduces the problem.