micropython/micropython #17243: Wireguard support (maybe generic vpn)

hupka36, 2025-05-04:
### Description

Hi, 

I have idea to use wireguard in very light style like : 

wg.start(private_key, peer_pubkey, endpoint_ip, endpoint_port)
wg.is_connected()
wg.send(data)
wg.recv()
...

but requirements for this is support for : 
WireGuard as an external native module, using MicroPython’s dynamic native modules support (.mpy with native code or user C module with dynamic linking).

then I can include WireGuard.mpy (creates resp. modified fork from ciniml/WireGuard-ESP32-Arduino)
- Uses ESP32 hardware crypto (via esp32-crypto) for speed.
- Implements WireGuard protocol in userspace (no kernel module needed).
- Designed for static peer-to-peer setups — perfect for embedded use.
- Can be reduced to single peer, single route, matching lightweight requirements.
(suitable also for ESP32-PICO or other)

If you have some idea?? Exist some way how to do that? Do you have better way how to create VPN client inside ESP32 (to be compatibile with mikrotik on other end of tunnel)???

Thank You very much for any response.

WBR -s-

### Code Size

It is space for discussion which way is correctly possible in micropython evolution.

### Implementation

I would like to sponsor development of this feature

### Code of Conduct

Yes, I agree

jonnor, 2025-05-31:
Hi @hupka36

Yes, this should be implementable as an external C module - it can then be included in a firmware build using USER_C_MODULES extension point. It likely is not suitable for a dynamic native module because it will need to integrate with the IP stack.

See https://github.com/cnadler86/micropython-camera-API for example of a recent external C module, as a possible starting point.

jonnor, 2025-05-31:
The implementation you refer to https://github.com/ciniml/WireGuard-ESP32-Arduino?tab=readme-ov-file is implemented using https://github.com/smartalock/wireguard-lwip
There seems to be another wrapper for esp-idf at https://github.com/trombik/esp_wireguard - which might be a better base.