micropython/micropython #17244: Example in library deflate documentation with deflate.DeflateIO and io.BytesIO throws OSError: [Errno 22] EINVAL

juhanijaakola, 2025-05-04:
### Port, board and/or hardware

ESP32_GENERIC_C3-20250415-v1.25.0

### MicroPython version

MicroPython v1.25.0 on 2025-04-15; ESP32C3 module with ESP32C3

### Reproduction

```
with open('response_body.bin', 'rb') as f:
    data = f.read()
len(data)
```
_1794_
```
import deflate, io
with deflate.DeflateIO(io.BytesIO(data), deflate.ZLIB) as d:
    decompressed_data = d.read()
```
_Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
OSError: [Errno 22] EINVAL_
```
with open('response_body.bin', 'rb') as f:
    with deflate.DeflateIO(f, deflate.GZIP) as d:
        decompressed_data = d.read()
```
_(Decompressing directly from a file stream works!)_
```
data[:40]
```
b'\x1f\x8b\x08\x00\x00\x00\x00\x00\x00\x03\xcd[\xdbn\xdbF\x10}\xcfW\x082\xfa\x12\xc4\xb2HK\x96\xec\x17c\x11\xb1\xf5\x9aWh\x97L'

### Expected behaviour

Expected to work without the EINVAL exception

### Observed behaviour

Using deflate.DeflateIO with a stream from io.BytesIO throws exception EINVAL

### Additional Information

Here is what the Linux file command says about the data to be decompressed:

$ file response_body.bin
response_body.bin: gzip compressed data, from Unix, original size modulo 2^32 15437

### Code of Conduct

Yes, I agree

juhanijaakola, 2025-05-04:
According to an example from [https://docs.micropython.org/en/latest/library/deflate.html](https://docs.micropython.org/en/latest/library/deflate.html#examples) **deflate.DeflateIO(io.BytesIO(compressed_data), deflate.ZLIB)** should work

MilhouseVH, 2025-05-05:
If the compressed data is GZIP format, does the following work?

```
import deflate, io
with deflate.DeflateIO(io.BytesIO(data), deflate.GZIP) as d:
    decompressed_data = d.read()
```

juhanijaakola, 2025-05-05:
The compressed data is in GZIP format, and deflate.GZIP works! Thanks!

closed 2025-05-05 by juhanijaakola