micropython/micropython #17245: USER_C_MODULE impossible to import library

dobodu, 2025-05-04:
### Port, board and/or hardware

ESP32S3 (16MB of Flash & Octal-SPIRAM)

### MicroPython version

MicroPython v1.26.0-preview.59.g79abdad9e.dirty on 2025-05-04
Build with ESP-IDF 5.4.1 toolchain

Hi, I've been upgraded Micropython to 1.26 and ESP-IDF to 5.4.1 since a bug report in my own repository.

I compile the firmware a usual (it was working with micropython 1.25 and ESP-IDF 5.3.2) without issue, However I cannot import the module as before. (import xxx lead to an ImportError: no module named 'xxx')

The help('modules') command shows usual library but not the one from the User_C_module (before I used to see xxx module)

The build command was 

make  BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=FLASH_16M_SPIRAM_OCT USER_C_MODULE=~/pico/XXX/xxx

I thought Issue #16424 would have closed the issue but no luck...

I'm not sure it's a bug, but I does look like...

### Reproduction

make  BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=FLASH_16M_SPIRAM_OCT USER_C_MODULE=~/pico/XXX/xxx

Was working well with  Micropython v1.25.0-preview.180.g495ce91ca.dirty on 2025-02-15  & IDF 5.3.2
Was not working  with  Micropython v1.26.0-preview.59.g79abdad9e.dirty on 2025-02-15  & IDF 5.3.2

So I expect it is not related to IDF version

### Expected behaviour

Normal output is 

import xxx ==> Nothing (OK)

### Observed behaviour

Observed output is 

import xxx ==> ImportError: no module named 'xxx'

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

dobodu, 2025-05-08:
Just a few more words 

I checked to micropython.cmake and the micropython.mk from my library => Seems OK
I checked the main xxx.c file to ensure that the module registration is correct (MP_QSTR_xxx ... MODULE_xxx_ENABLE)

I upgraded the build command line to :

make  BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=FLASH_16M_SPIRAM_OCT USER_C_MODULE=~/pico/[....]/xxx CFLAGS_EXTRA=-DMODULE_xxx=1 all

But doesn't solve the issue

I think the build process totally bypasses the compilation of the module (I thought it might be a module registration issue) but I don't see any output lines concerning the module compilation.... weird.

dobodu, 2025-05-08:
Still investigation.

Things are the same when I compile for PICO2_W the [st7789](https://github.com/russhughes/st7789_mpy) module

So problem seems not to be related with ESP-IDF.

dobodu, 2025-05-08:
When an S forces you to look like an idiot in face of the whole community 🥇 

USER_C_MODULE=~/pico/XXX/xxx  ==> USER_C_MODULE**S**=~/pico/XXX/xxx

Really sorry, Guys

closed 2025-05-08 by dobodu

Josverl, 2025-05-08:
I've made my share of typos, as I suppose anyone that has developed.

So no need to be embarrassed