micropython/micropython #17279: declaring an "big buffer" on esp32-s3 (16MB Flash & 8Mb RAM) need "long" time to recover connection after crash

chf-6309, 2025-05-10:
### Port, board and/or hardware

esp32-s3 "generic" (from AliExpress) YD-ESP32-23 / 2022-v1.3

### MicroPython version

MicroPython v1.25.0 on 2025-04-15; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3

### Reproduction

```
#LCD TFT size
width = 320
height = 480

# === Create a framebuffer RGB888 (3 bytes per pixel) ===
def create_framebuffer(width, height):
    return bytearray(width * height * 3)  # 460800 bytes - 461KB
```

called from main program:
`buf = create_framebuffer(width, height)`

### Expected behaviour

with 8MB available RAM, expect that, even is the buffer isn't used, the access to the esp32 card will be possible.
The size of that buffer should be around 460KB, small compared to the 8MB available

When my program crash during tests, after the buffer allowance, it need minutes or more, to recover the USB link to the card, or before I include an:
```
del buf
gc.collect()
```
at the end of the program.

The fastest way is to do an hard RESET

### Observed behaviour

the esp32 need minutes (more than five) before communication with PC and (portable) Thonny v4.17 is (again) possible, without the use of the hard reset.

First I didn't use the "Generic ESP32S3 module with **Octal-SPIRAM** with ESP32S3" firmware, so, the available RAM was little as ~300KB. Withe the Octal version I get;
```
Mémoire libre : **8312704 octets** [free memory]
stack: 736 out of 15360
GC: total: 64000, used: 8832, free: 55168, max new split: 8257536
 No. of 1-blocks: 176, 2-blocks: 22, max blk sz: 32, max free sz: 3410

MicroPython v1.25.0 on 2025-04-15; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3

Type "help()" for more information.

>>> import gc
>>> gc.mem_alloc()
**10640**
>>> gc.mem_free()
**7785024**
```

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

Josverl, 2025-05-10:
When using large buffers, the best thing is to allocate them **first**, before any other activities.

You did not share what other things your main.py does, so it's not clear at what stage you allocate.
You describe issues with network connectivity, but you have not shared how and when that is setup.

It would be helpful if you can share a minimal reproduction, that has the issue. I usually do this by commenting out chunks of code to find the root cause.
Adding prints / logging / turning on leds at certain stages also helps getting insight in what is happening on the board while you wait for reconnection.

chf-6309, 2025-05-10:
Hello Josverl,

thank you for your answer.
some more precision, as you ask:

- the size of my"(frame)buffer" is dynamic, i I need full screen (320*480*3 [one byte for each color]), some time I reduce the size, if I have to display only a small band of pixel on the display.
- on my tests, no **network** is used, but I mean the USB connection to the board (an USB to serial), sorry for the confusion, so, when a large buffer (max size ~600KB), even if I stop the program with a "CTRL+C", it seams that the program stop, give the hand... but is freezed for long minutes... or back after reset and Thonne IDE re-connection.
- here the content of the main program:
```
# main start
# hardware configuration
rst = machine.Pin(5)           # reset
cs_ili9488 = machine.Pin(2)    # chip select, for ili9488 module 3.50"
dc = machine.Pin(7)            # data/command for display - common

hspi = machine.SPI(1, baudrate=20000000, bits=8, sck=machine.Pin(14), mosi=machine.Pin(13), miso=machine.Pin(12))
hspi.init()            # Ensure right baudrate
display = ILI9488(hspi, cs=cs_ili9488, dc=dc, rst=rst)

# ili9488 initialyzation
print('init display')
display.init_display()

print('clear display')
display.cls()                          #use direct command to the ili9488
time.sleep_ms(500)
print('green background')
display.cls(0, 255, 0)
time.sleep_ms(500)

gc.collect()

buf = create_framebuffer(width, height) # full screen buffer ~600KB
#buf = create_framebuffer(128, 128)

print('red background using framebuffer') # check gain of speed with framebuffer
color = bytes([255, 0, 0])                # fsetup red color
buf[:] = color * (len(buf) // 3)          # fille the buffer with red color
display.draw_bitmap(0, 0, width, height, buf) # write buffer to ili9488
time.sleep_ms(50)                             # some time, and do the others colors
print('green background using framebuffer')
color = bytes([0, 255, 0])                # green
buf[:] = color * (len(buf) // 3)
display.draw_bitmap(0, 0, width, height, buf)
time.sleep_ms(50)
print('blue background using framebuffer')
color = bytes([0, 0, 255])                # blue
buf[:] = color * (len(buf) // 3)
display.draw_bitmap(0, 0, width, height, buf)
time.sleep_ms(50)

print('display image using framebuffer')
bufimg = bytearray(width * height * 3)        # setup to read a picture
load_bin_to_buffer('homerRGB888.bin', bufimg, 460804) # load picture
display.draw_bitmap(0, 0, width, height, bufimg)      # xfert picture buffer to screen
del bufimg                                    # delet one time image buffer
gc.collect()
time.sleep_ms(2000)

fill_buffer(buf)                              # black background by default
display.cls()                                 # clear screen

gc.collect()

# draw messages at different color/size on the buffer
draw_text_to_buffer(buf, width, 10, 30, "uPython ESP32-S3", 0, 255, 0, scale=2)
display.draw_bitmap(0, 0, width, height, buf)
draw_text_to_buffer(buf, width, 10, 54, "Size 3 test", 0, 0, 255, scale=3)
display.draw_bitmap(0, 0, width, height, buf)
draw_text_to_buffer(buf, width, 10, 86, "Size 4", 255, 0, 0, scale=4)
# then copy the buffer to the ili9488 (and LCD)
display.draw_bitmap(0, 0, width, height, buf)

del buf                                       # no more buffer use 
gc.collect()

for i in range (width):
    y=300+int(128*math.sin(2*math.radians(i)))
    #print(y)
    display.draw_pixel(i, y, 0, 255, 0)       # green sinusoid on display / no frame buffer

buf = create_framebuffer(width, 16)           # create a small framebuffer for 320x16 pixels
t = rtc.datetime()                            # read inernal RTC time
print ('Date: {:02d}/{:02d}/{:04d}_{:02d}:{:02d}:{:02d} * temperature: {:02d}°C'.format(t[2], t[1], t[0], t[4]+5, t[5], t[6], (esp32.mcu_temperature())))

while True:
    tx=0
    ty=12       # time position
    t = rtc.datetime()
    fill_buffer(buf)                         # fill buffer with "0,0,0" color, black
    draw_text_to_buffer(buf, width, 0, 0, 'Date: {:02d}/{:02d}/{:04d}_{:02d}:{:02d}:{:02d} * temp.: {:02d}°C'.format(t[2], t[1], t[0], t[4]+5, t[5], t[6], (esp32.mcu_temperature())), 0, 255, 0, scale=1)
    display.draw_bitmap(tx, ty, width, ty-height, buf) #xfert bramebuffer to LCD controler (320x16)
    time.sleep_ms(980)

del buf
gc.collect()
```
On the display class: (writes to the LCD controler)
```
def draw_bitmap(self, x, y, width, height, buf):
        self.write_cmd(0x2A)  # Column Address Set "x"
        self.write_data(bytearray([ x >> 8, x & 0xFF, (x + width - 1) >> 8, (x + width - 1) & 0xFF ]))
        self.write_cmd(0x2B)  # Page Address Set
        self.write_data(bytearray([ y >> 8, y & 0xFF, (y + height - 1) >> 8, (y + height - 1) & 0xFF ]))
        self.write_cmd(0x2C)  # Memory Write
        self.write_data(buf)
```

The things are working, but if I let that "big" buffer on memory, to have access to the esp32 throught the USB/serial connection, even after an CTRL-C stop, 

I hope that those information will help