micropython/micropython #17297: In the machine library and/or the pyb library, add a non-blocking UART transmit polling function

etherbrat, 2025-05-13:
### Description

STM32 UART.c inlines (via UART.h) the testing of the UART tx available bit and does not make the result publically available. This by definition eliminates the simplest way to implement cooperative multitasking, viz. spinning around a collection of non-blocking functions. Ironically, the machine library includes a POLL_HOOK function which is used by asyncio - however running up the asyncio learning curve is a high price to pay for a simple requirement. Given that the rx_any function has been provided for a non-blocking UART receive, it seems a bizarre oversight not to have provided the equivalent function for UART transmit.

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

robert-hh, 2025-05-13:
The STM port does not implement buffered transmit for UART. uart.write() is blocking until the last byte of the message is loaded into the UART. uart.flush(), uart.txdone() and the related ioctl() functions  are available for consistency. Implementing buffered TX is known to be missing and on my to-do list since 9/22, but I do not have many different boards, and so my confidence about test coverage is low.

etherbrat, 2025-05-13:
Ok, and maybe this doesn’t meet quality guidelines, but unknowns can always be covered off in the documentation.

peterhinch, 2025-05-13:
> running up the asyncio learning curve is a high price to pay for a simple requirement.

Nonblocking UART I/O with `asyncio` is straightforward. See [this example](https://github.com/peterhinch/micropython-async/blob/master/v3/as_demos/auart.py).

etherbrat, 2025-05-13:
Agreed, but it’s not the kind of solution that a user will be looking for, or anticipating.

dpgeorge, 2025-05-14:
@etherbrat what exactly are you asking for here?  There is `uart.txdone()`, or the more sophisticated `select.poll()` interface to work with non-blocking streams.

etherbrat, 2025-05-14:
Aside: MicroPython is an amazing product, as indicated by its rapid adoption on so many platforms. Much kudos!

As much as anything it comes down to the documentation, and failing that, the code. I started out as simple as possible by trying to use the pyb library, but if I am correct, tx_done is not supported there (Could it be? It would make life so much easier for 70-year old, hesitant code manglers like me). I moved on to the machine library, and sure enough, tx_done gets a mention. But for the absence of doubt, I went to uart.c and uart.h - and -  tx_done is nowhere to be found! It must be in one of the other include files. Anyway, if you, Damien George, say it's what I need, then so be it. The absence of an example that uses it didn't help.

FWIW, during my travels I encountered the timeout parameter to some functions. This came out of nowhere, was unexplained, and immediately spelled "**blocking**". I think you actually intend the parameter to be primarily for internal use, where you seem to calculate an appropriate value based on the number of characters in the buffer. In any case, a few words, to say that a timeout of 0 means non-blocking, would be very helpful.

robert-hh, 2025-05-14:
>  but if I am correct, tx_done is not supported there (Could it be? 

You're right. tx_done() is not, but txdone().

https://docs.micropython.org/en/latest/library/machine.UART.html#machine.UART.txdone

etherbrat, 2025-05-14:
Ah (Blush!)

peterhinch, 2025-05-14:
> I encountered the timeout parameter to some functions

Functions like `uart.read(n)` block. The `timeout` and `timeout_char` constructor args set limits to the blocking time. If you use `asyncio` these should be disabled; the `asyncio.wait_for` method should be used if a timeout is required.