micropython/micropython #17353: Build fails for port/unix mips/mipsel

o-murphy, 2025-05-24:
### Port, board and/or hardware

unix (mips/mipsel)

### MicroPython version

The build process for the mipsel port of MicroPython failed with the following error:
mipsel-linux-gnu-gcc: error: build-standard/lib/libffi/out/lib/libffi.a: No such file or directory
make: *** [../../py/mkrules.mk:244: build-standard/micropython] Error 1
make: Leaving directory '/home/murphy/micropython/ports/unix'

### Reproduction

Trying to make for unix (mipsel) in standalone mode
mipsel-linux-gnu-gcc

### Observed behaviour

mipsel-linux-gnu-gcc: error: build-standard/lib/libffi/out/lib/libffi.a: No such file or directory
make: *** [../../py/mkrules.mk:244: build-standard/micropython] Error 1

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

agatti, 2025-05-28:
It looks like you don't have `libtool` installed (see `ports/unix/README.md`).

o-murphy, 2025-05-29:
> It looks like you don't have `libtool` installed (see `ports/unix/README.md`).

Thanks a lot! Successfully built and tested on qemu-mipsel
But have issue with executing on target platform
The target platform is Ingenic halley5 (mipsel) board with buildroot

On host machine
```bash
$ file ./micropython
./micropython: ELF 32-bit LSB executable, MIPS, MIPS32 rel2 version 1 (SYSV), dynamically linked, interpreter /lib/ld.so.1, BuildID[sha1]=c33c8e6bae3f97312b94620a311cc9fcd2a595e6, for GNU/Linux 3.2.0, with debug_info, not stripped
```

On target
```bash
./micropython
/bin/sh: ./micropython: not found

ls -l /lib* | grep ld
-rwxr-xr-x    1 root     root        177152 Mar 24 11:41 ld-2.29.so
lrwxrwxrwx    1 root     root            10 Nov 28  2023 ld-linux-mipsn8.so.1 -> ld-2.29.so

/lib/ld-2.29.so ./micropython
./micropython: error while loading shared libraries: ./micropython: cannot open shared object file: No such file or directory

/lib/ld-linux-mipsn8.so.1 ./micropython
./micropython: error while loading shared libraries: ./micropython: cannot open shared object file: No such file or directory
```

o-murphy, 2025-05-30:
> Edit

Why does standalone micropyhon build depends on system libraries?

agatti, 2025-05-30:
Oh, I didn't build it as a standalone version, it was just to prove a point - `file` reports the binary is dynamically linked so I thought it had dependencies on some low level library like vdso or something :)

Anyway, I noticed that QEMU's sysroot has a file called `/lib/ld.so.1` (same as mentioned in `file`'s output) - so I wonder whether changing the interpreter path into `/lib/ld-2.29.so` with patchelf is actually worth doing.  The only MIPSEL boards I have around are based on MT7688 for which I don't have a working build environment at the moment so I cannot test my hypothesis on a live MIPSEL system.

o-murphy, 2025-05-30:
> Oh, I didn't build it as a standalone version, it was just to prove a point - `file` reports the binary is dynamically linked so I thought it had dependencies on some low level library like vdso or [...]

Problem is some target unix/mipsel environments have not compatible system deps (qemu has one) and it should be linked static for standalone unix/mips unix/mipsel builds (like libffi for now). 

I use now this setup (based on tools/ci.sh for unix/mips) for standalone build and yes, qemu-mipsel has ld.so.1. But the resulting "standalone" build is not fully standalone for now.
https://github.com/o-murphy/micropython-mipsel
You can check build-mipsel.sh and ci.sh