micropython/micropython #17398: RP2350 Regression in USB speed to 100kB/s

dalibor-drgon, 2025-05-30:
### Port, board and/or hardware

rp2350 from ports/rp2

### MicroPython version

I am playing with MicroPython and rewriting small C code to micropython in conquest to find new platform that would be easier to use for newbies. We currently use internally Zephyr project for projects that are as simple as GPIO accessing and processor communication with a salt of TCP and USB high-level protocol handling, and as complex as PIO reconfiguration and higher-performance data reading from it. Now I am entering MicroPython (but without Zephyr - only ports/rp2).

Using CDCInterface, I implemented simple CDC device that reads data nonblockingly and writes blockingly and was about to test. In Zephyr, we got about 500kB/s (probably because the USB Device driver was using the Zephyr's Device class using FIFO + interrupts - it should go faster). Here in Micropython, we get about 100kB/s.

I found this issue https://github.com/micropython/micropython/issues/7479 that was talking few years ago about low performing USB stack. I distilled my use case down to simple `x = b'x'*10000; while True: usys.stdout.buffer.write(x)` as in the above issue, and it seems to give similar performance /106kBps/ as well (not sure what differences are there - the CDCInterface seems to use tinyusb's _xfer_ functions). 

Is there a way how to fix this easily?

### Reproduction

- I built micropython from the website (micropython-1.25.0), but ported it to W5500-EVB-PICO2 board (basically by combining file from W5-EVB-PICO and default PICO2 board - could create pull request). Should work the same way with current boards.

### Expected behaviour

```py
import usys
x = b'x'*10000;
while True:
    usys.stdout.buffer.write(x)
```

### Observed behaviour

100kB/s transmission speed over custom USB / usys.stdout.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

dalibor-drgon, 2025-05-30:
EDIT: Sorry, but I got code inside my main.py file which automatically started high frequency Timer. Without it, this sample code runs at 500 KB/s. Tested on clean RP2040, will re-check on Monday again on actual W5500 board and close this bug if the issue is not present.

Sorry for the inconvenience / lost time

Josverl, 2025-06-06:
@dalibor-drgon 
Is there any feedback from the test to share ?

dalibor-drgon, 2025-06-08:
@Josverl Looks like the issue is not present (got 500KB/s). Closing it. Thanks a lot

closed 2025-06-08 by dalibor-drgon