micropython/micropython #17450: unable to compile mpy-cross

quantrpeter, 2025-06-07:
### Port, board and/or hardware

esp32

### MicroPython version

CC ../py/modio.c
CC ../py/modmath.c
CC ../py/modcmath.c
CC ../py/modmicropython.c
CC ../py/modstruct.c
CC ../py/modsys.c
CC ../py/moderrno.c
CC ../py/modthread.c
CC ../py/vm.c
CC ../py/bc.c
CC ../py/showbc.c
CC ../py/repl.c
CC ../py/smallint.c
CC ../py/frozenmod.c
CC main.c
CC gccollect.c
CC ../shared/runtime/gchelper_generic.c
error: Interference usage of base pointer/frame pointer.
1 error generated.
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [../py/mkrules.mk:101: build/shared/runtime/gchelper_generic.o] Error 1

### Reproduction

-

### Expected behaviour

-

### Observed behaviour

-

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

Josverl, 2025-06-07:
This is very unlikely a bug, most likely a misconfiguration of your environment.
For such things it is better to ask a question in the discussions.

quantrpeter, 2025-06-08:
I fixed it by using docker to build it in mac, thanks

closed 2025-06-08 by quantrpeter