micropython/micropython #17451: WEACT Blackpill problem

quantrpeter, 2025-06-07:
### Port, board and/or hardware

weact blackpill

### MicroPython version

Hi
    I have used st-flash to flash the hex to weact black pill board successfully, the serial port appear and i can connect to micropython The problem is : I have to connect both usb cable and st-link, if i plug out the st-link, the serial port never appear. Why?

st-flash --format ihex write build-WEACT_F411_BLACKPILL/firmware.hex

thanks
Peter

### Reproduction

-

### Expected behaviour

-

### Observed behaviour

-

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

andrewleech, 2025-06-08:
Which version black pill do you have? 
What speed crystal does it have loaded?
Which version / variant firmware are you trying to flash?

Josverl, 2025-06-08:
I think this would be better handled as a question in Discussios, rather than as a bug report