micropython/micropython #17486: Support for ESP32-C5

h-milz, 2025-06-13:
### Description

Since the ESP32-C5 is in mass production now and the Espressif boards can be officially purchased, I think it's time for a corresponding board description. A prerequisite is MPY support for ESP-IDF 5.5, though, which is due on June 30th. I have two ESP32-C5 sample boards lying around and could provide a ESP_GENERIC_C5 thing including code to support the 5 GHz wifi, 11ax and stuff. Anyone interested? 

### Code Size

a board description and a couple hundred bytes to include the ESP-IDF functions for 5 GHz wifi. 

### Implementation

I intend to implement this feature and would submit a Pull Request if desirable

### Code of Conduct

Yes, I agree

devick, 2025-07-15:
I would be interested!

Jibun-no-Kage, 2025-08-03:
Very much interested!  Need that 5GHz support!

jonnor, 2025-08-08:
ESP-IDF 5.5 was released some weeks ago https://github.com/espressif/esp-idf/releases/tag/v5.5
I think that means it would be possible for someone to start working on this. See existing ESP32 PRs for what is usually involved

h-milz, 2025-08-16:
> ESP-IDF 5.5 was released some weeks ago https://github.com/espressif/esp-idf/releases/tag/v5.5 I think that means it would be possible for someone to start working on this. See existing ESP32 PRs for what is usually involved

Currently MicroPython supports v5.2, v5.2.2, v5.3, v5.4, v5.4.1 and v5.4.2. I don't feel like trying to use 5.5 right now.

dpgeorge, 2025-08-16:
MicroPython does now support IDF v5.5, see commit 1273751a3b6f8a14679378e2e082e03979cb2ea8

dpgeorge, 2025-08-16:
And see #17912 for a PR for ESP32-C5 support.

agatti, 2026-03-16:
C5 support was added in 1.27.0, so I guess this can be closed?