micropython/micropython #17489: mpremote mip install uses /rom as default target.

robert-hh, 2025-06-13:
### Port, board and/or hardware

RP2 and others with vfsrom enabled

### MicroPython version

All versions with vfsrom support.

### Reproduction

1. Get or create a firmware with support for vfsrom. I used an RP2 board.
2. Populate the /rom directory
3. Call `mpremote mip install <filename>. In my example I used `neopixel` as file.

### Expected behaviour

Install the file to the board.

### Observed behaviour

The install crashes with an error message. The head of that message is:
```
 mpr mip install neopixel
Install neopixel
Installing neopixel (latest) from https://micropython.org/pi/v2 to /rom/lib
Installing: /rom/lib/neopixel.mpy
Traceback (most recent call last):
  File "/usr/local/bin/mpremote", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/robert/.local/lib/python3.11/site-packages/mpremote/main.py", line 569, in main
    handler_func(state, args)
  File "/home/robert/.local/lib/python3.11/site-packages/mpremote/mip.py", line 190, in do_mip
```
Using the --target option of mpremote avoids that problem. But anyhow /rom should not be accepable as target.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

dpgeorge, 2025-06-13:
I'm pretty sure this was fixed by 2db0c0225f41292f85b2cf49257d663856d5c8db .  Do you have the latest version of `mpremote`?

robert-hh, 2025-06-13:
Maybe not. I have to check it.

robert-hh, 2025-06-13:
I had `mpremote 1.25.0rc0.post369+gfac1a2463` installed. A fresh install of `mpremote 1.25.0` works.