micropython/micropython #17560: ESP32-S3 `mpremote` sends corrupted fs hook when mounting local directory

adzierzanowski, 2025-06-25:
### Port, board and/or hardware

ESP32-S3 N8R8

### MicroPython version

MicroPython 6fee099ca-dirty on 2025-06-25; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3

Compiled using
* IDF 5.4.1
* python 3.13.1
* gcc-15
* macOS Sonoma 15.5

`mpremote` is used directly from the micropython repository (but the results are same as with latest PyPI version)

### Reproduction

1. Have a local folder named `local`
2. Run `mpremote mount local`

### Expected behaviour

Expected to mount local directory on the device and enter the REPL

### Observed behaviour

```python
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File ".../micropython/tools/mpremote/mpremote/__main__.py", line 6, in <module>
    sys.exit(main.main())
             ~~~~~~~~~^^
  File ".../micropython/tools/mpremote/mpremote/main.py", line 615, in main
    handler_func(state, args)
    ~~~~~~~~~~~~^^^^^^^^^^^^^
  File ".../micropython/tools/mpremote/mpremote/commands.py", line 521, in do_mount
    state.transport.mount_local(path, unsafe_links=args.unsafe_links)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File ".../micropython/tools/mpremote/mpremote/transport_serial.py", line 304, in mount_local
    self.exec(fs_hook_code)
    ~~~~~~~~~^^^^^^^^^^^^^^
  File ".../micropython/tools/mpremote/mpremote/transport_serial.py", line 293, in exec
    raise TransportExecError(ret, ret_err.decode())
mpremote.transport.TransportExecError: Traceback (most recent call last):
  File "<stdin>", line 11
SyntaxError: invalid syntax
```

### Additional Information

After some digging I found a recent thread #17465 which helped to resolve the issue.

After changing the size of a data chunk sent at once over REPL, mounting works as expected:

```python
        for i in range(0, len(command_bytes), 32):
            self.serial.write(
                command_bytes[i : min(i + 32, len(command_bytes))]
            )
            time.sleep(0.01)
```

The `mpremote mount` command works with an earlier build I found on the internet ([this one](https://github.com/unixb0y/Micropython-ESP32-S3-WROOM-1-N4))

So I guess the real problem lies in some build flag.

### Code of Conduct

Yes, I agree

bixb922, 2025-06-29:
It seems the problem appears with commit d737112 "esp32/esp32_common.cmake: Use the tinyusb source files from ESP-IDF"

With the latest version of MicroPython v1.26.0-preview.277.g4bd99260d (2025-06-26) the problem is present.

I compiled a MP image including up to commit d737112 (fails) and then including up to the previous commit f77fd62  (works ok).

Here is the result with a image up to commit d737112 (fails):

```
MicroPython v1.26.0-preview.96.gd7371124d on 2025-06-29; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.
>>> import platform
>>> platform.platform()
'MicroPython-1.26.0-preview-xtensa-IDFv5.2.2-with-newlib4.3.0'
>>> 
% mpremote cp binary.10000 :/binary8.10000
cp binary.10000 :/binary8.10000
 ... copying  15% [###-----------------]mpremote: Error with transport:
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
```

Here is the result with a image up to commit f77fd62 (works ok):
```
MicroPython v1.26.0-preview.95.gf77fd6257 on 2025-06-29; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.
>>> import platform
>>> platform.platform()
'MicroPython-1.26.0-preview-xtensa-IDFv5.2.2-with-newlib4.3.0'
>>> 
% mpremote cp binary.10000 :/binary6.10000
cp binary.10000 :/binary6.10000
%
```

bixb922, 2025-07-02:
That change to ```idk_component.yml``` allows transfers. However, in average every 10 or so transfers crash with: ```OSError: [Errno 6] Device not configured```

To ensure a clean build, I started with empty folders. I installed latest MicroPython and installed ESP-IDF with:
```
% git clone -b v5.4.1 --recursive https://github.com/espressif/esp-idf.git
% cd esp-idf
% ./install.sh esp32s3
```
I made the change to the yaml file, but enclosed the version in quotes (I believe that doesn't change things):
```diff
% diff --git a/ports/esp32/main/idf_component.yml b/ports/esp32/main/idf_component.yml
index f7773f4f4..e0c60bfb0 100644
--- a/ports/esp32/main/idf_component.yml
+++ b/ports/esp32/main/idf_component.yml
@@ -4,7 +4,8 @@ dependencies:
   espressif/esp_tinyusb:
     rules:
       - if: "target in [esp32s2, esp32s3]"
-    version: "~1.0.0"
+    version: "^1.7.6"
+  espressif/tinyusb: "^0.18.0~2"
   espressif/lan867x:
     version: "~1.0.0"
     rules:
```
The ```make``` output shows that the required versions are processed:
```
NOTICE: Processing 4 dependencies:
NOTICE: [1/4] espressif/esp_tinyusb (1.7.6)
NOTICE: [2/4] espressif/mdns (1.1.0)
NOTICE: [3/4] espressif/tinyusb (0.18.0~2)
NOTICE: [4/4] idf (5.4.1)
```

When copying the same binary file the "Error in transport" does not appear any more, but after several transfers, a new error appearsr:
```
MicroPython v1.26.0-preview.290.g1b0cdc079.dirty on 2025-07-01; Generic ESP32S3 module with Octal-SPIRAM with ESP32S3
Type "help()" for more information.
>>> import platform
>>> platform.platform()
'MicroPython-1.26.0-preview-xtensa-IDFv5.4.1-with-newlib4.3.0'
>>>
% mpremote cp binary.10000 :/binary.10000
cp binary.10000 :/binary.10000
% mpremote cp binary.10000 :/binary.10001
cp binary.10000 :/binary.10001
% mpremote cp binary.10000 :/binary.10002
cp binary.10000 :/binary.10002
% mpremote cp binary.10000 :/binary.10003
cp binary.10000 :/binary.10003
% mpremote cp binary.10000 :/binary.10004
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.12/bin/mpremote", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/mpremote/main.py", line 569, in main
    handler_func(state, args)
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/mpremote/commands.py", line 327, in do_filesystem
    state.ensure_raw_repl()
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/mpremote/main.py", line 511, in ensure_raw_repl
    self.transport.enter_raw_repl(soft_reset=soft_reset)
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/mpremote/transport_serial.py", line 171, in enter_raw_repl
    data = self.read_until(1, b"raw REPL; CTRL-B to exit\r\n", timeout_overall=timeout_overall)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/mpremote/transport_serial.py", line 123, in read_until
    elif self.serial.inWaiting() > 0:
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/serial/serialutil.py", line 594, in inWaiting
    return self.in_waiting
           ^^^^^^^^^^^^^^^
  File "/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/serial/serialposix.py", line 549, in in_waiting
    s = fcntl.ioctl(self.fd, TIOCINQ, TIOCM_zero_str)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 6] Device not configured
```

I don't see this behavior on the same Mac, same cables, same microcontroller,  but with commit d737112 excluded.

agatti, 2025-07-02:
Thanks for trying this out - the quotes aren't really necessary, as it's a YAML file anyway :)

This is interesting, consider that the version of TinyUSB that MicroPython brings in as a submodule is actually *older* than the one you had issues before and fewer issues now.

So this may mean two things: Espressif's vendored TinyUSB 0.18 has issues, or its integration has issues.  You've tried different versions of `espressif/esp_tinyusb` but turns out that it defaults to the latest available version of `espressif/tinyusb` anyway, or at least it does here:

```
...
NOTICE: Updating lock file at /home/agatti/src/micropython/ports/esp32/dependencies.lock
NOTICE: Processing 4 dependencies:
NOTICE: [1/4] espressif/esp_tinyusb (1.0.3)
NOTICE: [2/4] espressif/mdns (1.1.0)
NOTICE: [3/4] espressif/tinyusb (0.18.0~2)
NOTICE: [4/4] idf (5.4.0)
...
```

The only meaningful changes in the Espressif's version of `espressif/tinyusb` you tested compared to the official release are the addition of a header file that is used by `espressif/esp_tinyusb` and a CMake build script that sets `CFG_TUSB_MCU` to `esp32sx` for your specific use case.

Given that MicroPython vendors version 0.17.0 of TinyUSB, if it's not too much to ask, would it be possible repeat the test but leaving `espressif/esp_tinyusb`'s version alone and using version `0.17.0~2` or `0.17.0~1` of espressif/tinyusb?

That would help ruling out whether it's just the slightly patched TinyUSB or if it's Espressif's integration at fault here, before I send a patch that reverts my change.

agatti, 2025-07-03:
Thanks again for testing those changes.

I'd say that it looks like TinyUSB's `espressif/esp32sx` driver has issues with macOS (or the other way around, actually, given that Linux and Windows don't seem to mind), whilst MicroPython used the `synopsis/dwc2` driver instead.

I can revert d7371124d288cc85d2386b63ea39cfa4fa803194, but I realise that it'd break MicroPython as an ESP-IDF component since their CMake scripts - at least on 5.2/5.2.2 and on 5.3 too if I remember correctly - freak out when components' source trees change; `tinyusb` is fetched as a submodule and there's no way to provide an alternate location for that subrepo at the moment (in fact `micropython-lib` has to come from somewhere else too and its path passed as a CMake variable).  

Fun times...  If anything then tinyusb will need to have its own root path variable too.

@andrewleech when you did the original tinyusb integration, was it there a specific reason to pick `dwc2` rather than `esp32sx`?

andrewleech, 2025-07-03:
>
> [@andrewleech](https://github.com/andrewleech) when you did the original tinyusb integration, was it there a specific reason to pick `dwc2` rather than `esp32sx`?

I really just picked `dwc2` beacuse it was said to work according to tinyusb docs and I considered it more up-to-date / better tested because it's used on such a wide range of ports.

It sounds like if there's a bug in the espressif driver then it should be reported to them, though I'm not sure how to describe / reproduce it outside micropython which might make it harder for them to diagnose. Or maybe we should just propose they have a compile option to select which tinyusb driver is used by their component?

agatti, 2025-07-03:
Right, makes sense!  I also agree about how tricky would be to provide a test case to Espressif for this specific issue.

Maybe it'd help to reproduce the "device not configured" issue using the latest version of the `espressif/esp_tinyusb` component, with a custom built firmware having ESP-IDF's verbosity set to maximum.  The trick is to not only crank up the verbosity but also to get the log messages on either a different UART or over the JTAG port (assuming the extra logs on the UART would confuse mpremote).  A different UART connected via a USB-UART TTL adapter would be the best case, since if JTAG connectivity over USB gets cut short due to a driver failure, you may miss some error messages.

Maybe it's something as simple as ESP-IDF running out of heap space after a while because macOS doesn't terminate the transfer correctly or something like that, given that the issue happens after several transfers in a row.

@bixb922 I have no idea on which is the correct configuration entries' incantation to get the log output somewhere else but UART0, and I don't have an ESP32S3 board available where I am right now.  However, increasing verbosity *should* be achieved with this (can't try until I get home, sorry):

```diff
diff --git i/ports/esp32/boards/sdkconfig.base w/ports/esp32/boards/sdkconfig.base
index 30740af43..4a3c46136 100644
--- i/ports/esp32/boards/sdkconfig.base
+++ w/ports/esp32/boards/sdkconfig.base
@@ -17,7 +17,8 @@ CONFIG_BOOTLOADER_SKIP_VALIDATE_IN_DEEP_SLEEP=y
 CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE=y
 
 # Change default log level to "ERROR" (instead of "INFO")
-CONFIG_LOG_DEFAULT_LEVEL_ERROR=y
+#CONFIG_LOG_DEFAULT_LEVEL_ERROR=y
+CONFIG_LOG_DEFAULT_LEVEL_VERBOSE=y
 
 # Set the maximum included log level higher than the default,
 # so esp.osdebug() can enable more logging at runtime.
@@ -26,9 +27,9 @@ CONFIG_LOG_DEFAULT_LEVEL_ERROR=y
 # CONFIG_LOG_MAXIMUM_LEVEL_INFO=y and uncomment one of the other settings.
 #
 # If not needed, the next line can be commented entirely to save binary size.
-CONFIG_LOG_MAXIMUM_LEVEL_INFO=y
+#CONFIG_LOG_MAXIMUM_LEVEL_INFO=y
 #CONFIG_LOG_MAXIMUM_LEVEL_DEBUG=y
-#CONFIG_LOG_MAXIMUM_LEVEL_VERBOSE=y
+CONFIG_LOG_MAXIMUM_LEVEL_VERBOSE=y
```

agatti, 2025-07-04:
Wow, thanks a lot for the stack trace!

I'd wager something else in the USB stack stops handling packets for some other reason and thus the internal USB packet queue gets full.  I admit I'm a bit out of my depth here, I also expected more logging messages coming from TinyUSB and ESP-IDF's USB integration.

From a quick look it might seem that the USB task in `espressif__tinyusb/src/device/usbd.c` (`tud_task_ext`) either crashed or got stuck.  If only I still had a mac :)

@andrewleech sorry for bothering you again but you might have seen something like this whilst writing the integration code...  Any ideas on how to go further from here as far as tinyusb goes?

andrewleech, 2025-07-05:
It's no bother I'm quite interested in what's going wrong here. I didn't see any issues like this no, I thought it all pretty much just worked once the basics were all in place.

agatti, 2025-07-05:
Interesting!  I noticed Espressif did think of a similar scenario, but for device disconnection: https://github.com/espressif/tinyusb/commit/d4a39f5a7bd7c613511f740c1103346e2056ee11 - this change is part of the TinyUSB source bundle being compiled in this case.

I wasn't able to find any useful resource about the USB interrupts firing during flash - I believe Claude misunderstood reports of "device flashing over USB was interrupted" with "USB interrupts firing during SPI flash operations".

Regarding `CFG_TUSB_OS`, `espressif/esp_tinyusb` always set that define to `OPT_OS_FREERTOS` even back when it used to work.  There's one reference to that define at the end of `tud_task_ext` where if it's running on FreeRTOS it would exit the reading loop once all pending events are processed.  However, if I read the code correctly in `shared/tinyusb/mp_usbd.c`, that should be its expected behaviour as an internal MicroPython task would keep calling that function at regular intervals as long as there's one or more USB devices connected.  So this may be another red herring.

What bothers me the most, though, is that given `espressif/esp_tinyusb` brings in `espressif/tinyusb`, and the original integration would also bring in files from a vendored TinyUSB source tree, why there were no duplicate symbols in the first place?

agatti, 2025-07-05:
I've also looked more into interrupt priority aspect.

From the stack trace, the USB interrupt handler is first invoked by `vPortClearInterruptMaskFromISR`, which in turn goes through the Xtensa vector handler, which calls the USB interrupt handler again, triggering the assert.

`vPortClearInterruptMaskFromISR` is supposed to re-enable interrupts from inside a nested interrupt handler: https://github.com/espressif/esp-idf/blob/5b11d5b26a8bf151fc6bac400158859eedd413bc/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h#L171 and `_xt_lowint1` is the *low priority* CPU interrupt handler: https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-guides/hlinterrupts.html#interrupt-priorities

Looking at some of the FreeRTOS internal code that appears in the stack trace past `spi_flash_op_unlock` it seems that `vPortClearInterruptMaskFromISR` re-enables the interrupts that were previously disabled by the critical section.  This means that SPI code is probably already running at a higher priority than the USB stack, so lowering USB's interrupt priority isn't much of a solution.

It may possibly be that an USB interrupt is fired *while* the interrupt function wrapper is running, as the code is outside an interrupt context, otherwise it's not clear how exiting a critical section would call into an interrupt handler directly without going through the CPU's vectors.

The question is how the (wrapped) interrupt handler gets called the first time, but maybe there's some stack trickery going on that may affect the stack trace as it is effectively just a glorified memory area dump after all...

agatti, 2025-07-19:
Hrm, I'm having a similar issue with the version of TinyUSB that works on the ESP32-S3, but on a different board/architecture/target OS/connecting OS.  Not sure if it's a complete coincidence or it is something going on with MicroPython's shared integration bits?

The stack trace on my board is as such, just in case (IRQ handlers calling each other is how GCC decided the same implementation for multiple IRQs had to be done, go figure):

```
01:35:19 queue_event > bt
#0  queue_event (in_isr=<optimized out>, event=<optimized out>) at ../../lib/tinyusb/src/device/usbd.c:346
#1  dcd_event_handler (event=<optimized out>, in_isr=<optimized out>)
    at ../../lib/tinyusb/src/device/usbd.c:1214
#2  0x0001bcbe in dcd_event_xfer_complete (in_isr=true, result=0 '\000', xferred_bytes=<optimized out>, 
    ep_addr=<optimized out>, rhport=0 '\000') at ../../lib/tinyusb/src/device/dcd.h:224
#3  handle_ctr_tx (ep_id=0) at ../../lib/tinyusb/src/portable/st/stm32_fsdev/dcd_stm32_fsdev.c:297
#4  dcd_int_handler (rhport=rhport@entry=0 '\000')
    at ../../lib/tinyusb/src/portable/st/stm32_fsdev/dcd_stm32_fsdev.c:455
#5  0x00018b04 in USBFS_IRQHandler () at usb.c:125
#6  0x00018b9c in USB_LP_CAN1_RX0_IRQHandler () at usb.c:132
#7  0x00000000 in ?? ()
```

Fun fact, this is not a STM32-based board (it's not even Arm based), but the chip manufacturer loved STM so much they've basically copied the USB stack's register names and behaviour, so TinyUSB's STM32 implementation works on a random Chinese RISC-V device as well.

pavelrevak, 2025-09-01:
see similar problems,

mpremote is not able to upload files. the uploading fails during upload or uploaded files are corrupted.

my configuration, where I see this issue is:
- MacOS + ESP32-S3 + octal-PSRAM (XIAO-S3)

without issues:
- other boards connected to MacOS (C3, C6, ..)
- running mpremote on Linux

pavelrevak, 2025-09-01:
@dpgeorge thanks, #17960 fixes the issue.

PR #17960 (2025-09-10): esp32: Bump esp_tinyusb component, add component version lock files to source control.
### Summary

Three changes: two related to ESP-IDF dependencies, one to TinyUSB:

* Update the esp_tinyusb version to the latest to partly fix native USB serial corruption issues from macOS hosts:
  - Alternative to #17913.
  - Relates to https://github.com/orgs/micropython/discussions/17465.
  - Relates to https://github.com/micropython/micropython/issues/17865 but may not fix @gohai's issue.
  - Closes #17560 (probably).
* Rewrite the CDC write function to allow for the new `tx_overwritabe_if_not_connected [... cut]
closed 2025-09-10 by dpgeorge