micropython/micropython #17573: Can't not set  i2s  mck parameter esp32s3 MicroPython (1.25.0)

RonTang, 2025-06-27:
### Port, board and/or hardware

esp32s3

### MicroPython version

Can't not set  i2s  mck parameter esp32s3 MicroPython (1.25.0) 
my fireware is MicroPython v1.19.1-722-g1b774b373-dirty on 2022-11-29
here is my code:
audio_in = I2S(0,
sck=bck_pin, ws=ws_pin, sd=sdin_pin,mck=mck_pin,
mode=I2S.RX,
bits=16,
format=I2S.MONO,
rate=8000,
ibuf=64000)
TypeError: extra keyword arguments given
but if i del mck ,it works normally

### Reproduction

```python

my fireware is MicroPython v1.19.1-722-g1b774b373-dirty on 2022-11-29
here is my code:
audio_in = I2S(0,
sck=bck_pin, ws=ws_pin, sd=sdin_pin,mck=mck_pin,
mode=I2S.RX,
bits=16,
format=I2S.MONO,
rate=8000,
ibuf=64000)
TypeError: extra keyword arguments given
but if i del mck ,it works normally
like this:
audio_in = I2S(0,
sck=bck_pin, ws=ws_pin, sd=sdin_pin,
mode=I2S.RX,
bits=16,
format=I2S.MONO,
rate=8000,
ibuf=64000)
```

### Expected behaviour

no type error

### Observed behaviour

raise type error

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

ricksorensen, 2025-06-27:
See #17068.

Currently esp32/machine_i2s.c does not allow user setting of `mclk` and sets  `mclk` parameter to unused GPIO, 
```c
...
    i2s_std_config_t std_cfg = {
        .clk_cfg = I2S_STD_CLK_DEFAULT_CONFIG(self->rate),
        .slot_cfg = slot_cfg,
        .gpio_cfg = {
            .mclk = I2S_GPIO_UNUSED,
            .bclk = self->sck,
            .ws = self->ws,
            .invert_flags = {
                .mclk_inv = false,
                .bclk_inv = false,
                .ws_inv = false,
            },
        },
    };

    if (mode == MICROPY_PY_MACHINE_I2S_CONSTANT_TX) {
        std_cfg.gpio_cfg.dout = self->sd;
        std_cfg.gpio_cfg.din = I2S_GPIO_UNUSED;
    } else { // rx
        std_cfg.gpio_cfg.dout = I2S_GPIO_UNUSED;
        std_cfg.gpio_cfg.din = self->sd;
    }
...
```

RonTang, 2025-06-30:
> See [#17068](https://github.com/orgs/micropython/discussions/17068).
> 
> Currently esp32/machine_i2s.c does not allow user setting of `mclk` and sets `mclk` parameter to unused GPIO,
> 
> ...
>     i2s_std_config_t std_cfg = {
>         .clk_cfg = I2S_STD_CLK_DEFAULT_CONFIG(self->rate),
>         .slot_cfg = slot_cfg,
>         .gpio_cfg = {
>             .mclk = I2S_GPIO_UNUSED,
>             .bclk = self->sck,
>             .ws = self->ws,
>             .invert_flags = {
>                 .mclk_inv = false,
>                 .bclk_inv = false,
>                 .ws_inv = false,
>             },
>         },
>     };
> 
>     if (mode == MICROPY_PY_MACHINE_I2S_CONSTANT_TX) {
>         std_cfg.gpio_cfg.dout = self->sd;
>         std_cfg.gpio_cfg.din = I2S_GPIO_UNUSED;
>     } else { // rx
>         std_cfg.gpio_cfg.dout = I2S_GPIO_UNUSED;
>         std_cfg.gpio_cfg.din = self->sd;
>     }
> ...

Thanks for your reply. Hope to support custom mclk for esp32 as soon as possible！