micropython/micropython #17577: ESP32-C6 timers not working on custom build

AndreoBotelho, 2025-06-28:
### Port, board and/or hardware

esp32-c6 custorm pcb

### MicroPython version

MicroPython v1.25.0 on 2025-06-28; ESP32C6 module with ESP32C6

built with espidf v5.3.2 or 5.4.1, trying v1.26 preview form master branch gives same result

### Reproduction

1. create a timer ( 0 or 1)
2. init it and set a callback
3. wait for callback result

```
def mycallback(t):
    print('timer')

tim = Timer(0)

tim.init(period=100, callback=mycallback)
```

### Expected behaviour

Expected print "timer"

### Observed behaviour

nothing happens, working fine on espidf 5.2.5

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

AndreoBotelho, 2025-07-12:
Can someone help to find the cause?

andrewleech, 2025-07-13:
I don't suppose you've got any different esp32 chips you can test the same issue on? It might help to know if it's C6 specific. 

Also do you mean you downloaded the pre-built ESP32_GENERIC_C6 master branch binary and it has the same problem?

There's any number of changes in the IDF versions that might contribute to this.
It looks like there were changes to the higher level gptimer API between these versions that could be related, except micropython still looks like it's using an older lower level api, perhaps it's not fully compatible with the C6.

AndreoBotelho, 2025-07-13:
I've tested on C6 and C3, C3 is working fine, that is why I reported as C6 bug. It appears to be relates do LL api support on C6 but Espressif says this api may break anytime and its examples are note using it (esp-idf timer examples work fine), maybe the solution is to rewrite the timer support but if its is just C6 specific could have a workaroud.