micropython/micropython #17596: zephyr: Pin.value() always returns 0

bikeNomad, 2025-07-01:
### Port, board and/or hardware

zephyr, xiao_ble/nrf52840/sense

### MicroPython version

MicroPython v1.26.0-preview.290.g1b0cdc079 on 2025-07-01; zephyr-xiao_ble with nrf52840

My own board port, see https://github.com/bikeNomad/micropython-xiao_ble_nrf52840_sense/tree/main/zephyr

### Reproduction

Run the following in the REPL to reproduce:
```
from machine import Pin
p = Pin(('gpio0', 26), Pin.OUT, value=0) # red LED should go ON
p.value()  # answers 0
p.value(1) # red LED goes OFF
p.value() # answers 0 but should answer 1
```

### Expected behaviour

The red LED should go ON and then OFF (this works).
`p.value()` should answer `0` in the first case, and `1` in the second.

### Observed behaviour

`p.value()` answered `0` in both cases.

### Additional Information

I just built MicroPython for the XIAO BLE nrf52840 Sense board using Zephyr.
Here's my config and DTS files: https://github.com/bikeNomad/micropython-xiao_ble_nrf52840_sense/tree/main/zephyr

### Code of Conduct

Yes, I agree

dpgeorge, 2025-07-02:
I tested `bbc_microbit_v2` and indeed you cannot read an output value.

Should be fixed by #17599.

bikeNomad, 2025-07-02:
Fixed by #17599

closed 2025-07-02 by bikeNomad