micropython/micropython #17600: Add support for STM32WB6.

Vincent1-python, 2025-07-02:
### Description

Add support for STM32WB6, which has been out for a while.

### Code Size

uncertain

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

dpgeorge, 2025-07-02:
I haven't heard of the WB6, do you mean something else?  Can you link to the product page?

Vincent1-python, 2025-07-03:
https://www.st.com/en/evaluation-tools/nucleo-wba65ri.html 

https://www.st.com.cn/zh/microcontrollers-microprocessors/stm32wba65ri.html
Like these two

Message ID: ***@***.***>

Vincent1-python, 2025-07-03:
No, I use https://www.st.com/en/microcontrollers-microprocessors/stm32wba65ri.html

Message ID: ***@***.***>

Vincent1-python, 2025-07-03:
Now that I've finished correcting it, I can start.

dpgeorge, 2025-07-03:
OK, so this is the new STM32WBA line of MCUs.  The Cube package with CMSIS is here: https://github.com/STMicroelectronics/STM32CubeWBA

Vincent1-python, 2025-07-03:
Yes, so can it be realized? You can't see it until version 1.26, right?

Vincent1-python, 2025-07-03:
I think the sooner it comes out, the better. I really want to run MicroPython on it. The board has been on the table for a while and has fallen into dust. I want it to run quickly.

dpgeorge, 2025-07-03:
It will be possible to support this MCU, but it's likely a fair bit of work/effort, and so will take some time to do.

Vincent1-python, 2025-07-03:
Ok, I'll wait.

Vincent1-python, 2025-08-14:
Has the work begun.