micropython/micropython #17621: Raspberry Pico W: Hangs after WLAN deinit()

SmokeyBrandie, 2025-07-06:
### Port, board and/or hardware

pico w

### MicroPython version

MicroPython v1.25.0 on 2025-04-15; Raspberry Pi Pico W with RP2040

### Reproduction

```
import network
import time
from time import sleep
import urequests
import gc

WLAN_SSID = 'MYSSID'
WLAN_PASSWORD = '123'

wlan = network.WLAN(network.STA_IF)

counter = 1
while True:
    print('Start round {}'.format(counter))

    wlan.active(True)
    wlan.connect(WLAN_SSID, WLAN_PASSWORD)

    for i in range(20):
        if wlan.isconnected():
            break
        print('Trying to connect... Status is: ', wlan.status())
        time.sleep(1)

    if wlan.isconnected():
        response = urequests.get('https://www.google.com')
        response.close()

    print('Deactivate WLAN...')
    print('Memory alloc {}'.format(gc.mem_alloc()))
    print('Memory free {}'.format(gc.mem_free()))

    wlan.disconnect()
    wlan.active(False)
    wlan.deinit()

    print('Sleeping...')
    time.sleep(5)
    counter += 1
```

### Expected behaviour

Should not hang

### Observed behaviour

This script works approx. 10-30 times and then it suddenly hangs. It seems to be the 
deinit() function which causes the freeze.

I tested several pico w hardware, but same problem. 

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

agatti, 2025-07-06:
For people following along at home:

```
#0  HardFault_Handler () at /micropython/lib/pico-sdk/src/rp2_common/pico_crt0/crt0.S:170
#1  <signal handler called>
#2  mdns_add_query_question_to_outpacket (outpkt=outpkt@entry=0x20041d34, msg=msg@entry=0xac) at /micropython/lib/lwip/src/apps/mdns/mdns_out.c:559
#3  0x100450fa in mdns_create_outpacket (netif=netif@entry=0x20002e98 <cyw43_state+2276>, msg=msg@entry=0xac, outpkt=outpkt@entry=0x20041d34)
    at /micropython/lib/lwip/src/apps/mdns/mdns_out.c:588
#4  0x1004538e in mdns_send_outpacket (msg=msg@entry=0xac, netif=netif@entry=0x20002e98 <cyw43_state+2276>) at /micropython/lib/lwip/src/apps/mdns/mdns_out.c:770
#5  0x100455ba in mdns_multicast_probe_timeout_reset_ipv6 (arg=0x20002e98 <cyw43_state+2276>) at /micropython/lib/lwip/src/apps/mdns/mdns_out.c:978
#6  0x10054aae in sys_check_timeouts () at /micropython/lib/lwip/src/core/timeouts.c:390
#7  0x10040482 in mp_network_soft_timer_callback (self=<optimized out>) at /micropython/ports/rp2/mpnetworkport.c:139
#8  0x1003d45a in soft_timer_handler () at /micropython/shared/runtime/softtimer.c:93
#9  0x10040828 in PendSV_Handler () at /micropython/ports/rp2/pendsv.c:177
#10 <signal handler called>
#11 pendsv_resume_run_dispatch () at /micropython/ports/rp2/pendsv.c:123
#12 0x100407dc in pendsv_resume () at /micropython/ports/rp2/pendsv.c:79
#13 0x10040504 in lwip_lock_release () at /micropython/ports/rp2/mpnetworkport.c:133
#14 0x10065d22 in cyw43_deinit (self=0x200025b4 <cyw43_state>) at /micropython/lib/cyw43-driver/src/cyw43_ctrl.c:145
#15 0x1003269c in network_cyw43_deinit (self_in=<optimized out>) at /micropython/extmod/network_cyw43.c:135
#16 0x1001568a in fun_builtin_1_call (self_in=0x100a542c <network_cyw43_deinit_obj>, n_args=<optimized out>, n_kw=<optimized out>, args=0x20041ec8)
    at /micropython/py/objfun.c:70
#17 0x1002062e in mp_call_function_n_kw (fun_in=0x100a542c <network_cyw43_deinit_obj>, n_args=1, n_kw=n_kw@entry=0, args=args@entry=0x20041ec8)
    at /micropython/py/runtime.c:726
#18 0x1002071e in mp_call_method_n_kw (n_args=<optimized out>, n_kw=0, args=args@entry=0x20041ec4) at /micropython/py/runtime.c:742
#19 0x100236ae in mp_execute_bytecode (code_state=code_state@entry=0x20041eb0, inject_exc=inject_exc@entry=0x0) at /micropython/py/vm.c:1042
#20 0x100158a6 in fun_bc_call (self_in=0x1002062f <mp_call_function_n_kw+42>, n_args=<optimized out>, n_kw=0, args=0x0) at /micropython/py/objfun.c:295
#21 0x1002062e in mp_call_function_n_kw (fun_in=fun_in@entry=0x2000eff0, n_args=n_args@entry=0, n_kw=n_kw@entry=0, args=args@entry=0x0) at /micropython/py/runtime.c:726
#22 0x1002065c in mp_call_function_0 (fun=fun@entry=0x2000eff0) at /micropython/py/runtime.c:700
#23 0x1003cf88 in parse_compile_execute (source=source@entry=0x20041f84, input_kind=input_kind@entry=MP_PARSE_FILE_INPUT, exec_flags=exec_flags@entry=65)
    at /micropython/shared/runtime/pyexec.c:119
#24 0x1003d064 in do_reader_stdin (c=<optimized out>) at /micropython/shared/runtime/pyexec.c:276
#25 0x1003d0be in pyexec_raw_repl () at /micropython/shared/runtime/pyexec.c:509
#26 0x1003ff40 in main (argc=<optimized out>, argv=<optimized out>) at /micropython/ports/rp2/main.c:232
```

with `lib/lwip/src/apps/mdns/mdns_out.c` receiving an invalid pointer at line 556:

```c
22:59:10 mdns_add_query_question_to_outpacket > l
554       /* Write legacy query question */
555       if(msg->query) {
556         struct mdns_request *req = msg->query;
557         struct mdns_domain dom;
558         /* Build question domain */
559         mdns_build_request_domain(&dom, req, req->name[0]);
560         /* Add query question to output packet */
561         res = mdns_add_question(outpkt, &dom, req->qtype, DNS_RRCLASS_IN, 0);
562         if (res != ERR_OK) {
563           return res;
22:59:11 mdns_add_query_question_to_outpacket > print req
$1 = (struct mdns_request *) 0x1ca5a46
```

RAM starts at 0x2000_0000, Flash at 0x1000_0000.  The mDNS request points nowhere useful.

On my network it crashes on the second loop iteration, go figure.

andrewleech, 2025-07-06:
There's likely something along the lines of the netif being deregistered from lwip during deinit and some other parts of the stack services shut down but something there missing relating to mdns leaving it attempting to run here.

I've started a project to clean up NIC init /deinit /active handling starting with https://github.com/micropython/micropython/pull/17613 and https://github.com/micropython/micropython/pull/16459

If that goes well I'd like to roll those changes out across all the network adaptors to make their behaviour much more consistent.

SmokeyBrandie, 2025-07-12:
I could compile the micropython firmware with your bug fix. It works perfectly, thank you very much. Hopefully it will be merged into the official 1.26 release soon. Again, thank you all for the very fast and professional analysis and for the fix.

dpgeorge, 2025-07-17:
Fixed by 0b698b82417b17b1098af9e64433741154345252

closed 2025-07-17 by dpgeorge