micropython/micropython #17651: MICROPY_DEF_BOARD doesn't support parentheses in MICROPY_HW_BOARD_NAME

sfe-SparkFro, 2025-07-09:
### Port, board and/or hardware

rp2, though I suspected all other ports are affected too

### MicroPython version

MicroPython 05342b013-dirty on 2025-07-09; Raspberry Pi Pico with RP2040

### Reproduction

Create a variant for `RPI_PICO` called `mpconfigvariant_MY_VARIANT.cmake` with contents:

```
list(APPEND MICROPY_DEF_BOARD
    MICROPY_HW_BOARD_NAME="Raspberry Pi Pico (MY VARIANT)"
)
```

Build (`make BOARD=RPI_PICO BOARD_VARIANT=MY_VARIANT`) and flash to board.

### Expected behaviour

Normal board:

```
>>> import sys
>>> sys.implementation._machine
'Raspberry Pi Pico with RP2040'
```

My variant:

```
>>> import sys
>>> sys.implementation._machine
'Raspberry Pi Pico (MY VARIANT) with RP2040'
```

### Observed behaviour

With my variant, all contents of `MICROPY_HW_BOARD_NAME` are missing:

```
>>> import sys
>>> sys.implementation._machine
' with RP2040'
```

### Additional Information

If `MICROPY_HW_BOARD_NAME` is modified directly in `mpconfigboard.h` (not a variant), then it works fine:

```
>>> import sys
>>> sys.implementation._machine
'Raspberry Pi Pico (MY VARIANT) with RP2040'
```

### Code of Conduct

Yes, I agree

sfe-SparkFro, 2025-07-09:
To be clear, setting `MICROPY_HW_BOARD_NAME` to a value without parentheses works fine. It's just when parentheses are included that `MICROPY_HW_BOARD_NAME` is missing from `sys.implementation._machine`. I'm exploring making some boards variants for another project, and hoping to have variant names specified with parentheses if possible 😃

Josverl, 2025-07-10:
In current preview builds 
`sys.implementation._build` will show both the board and the variant 

Also, as the variant is used in the filename to store the pre built firmwares,   using () is asking for problems in unexpected places...