micropython/micropython #17667: Add USB Host

Vincent1-python, 2025-07-13:
### Description

After all, adding USB Host can be used on esp32p4.

### Code Size

None

### Implementation

I hope the MicroPython maintainers or community will implement this feature

### Code of Conduct

Yes, I agree

andrewleech, 2025-07-14:
I do have a micropython USB Host integration underway already based on the TinyUSB Host support. The initial pr on this will support CDC serial, MSC (mass storage) and HID. There's future potential to expand this to runtime / arbitrary device support but the API / format of that will require far more significant planning. 

I have this working on a branch already with all three classes mentioned above, however I think a little more fixup might still be required on cdc serial. Along with that I have a bunch of automated testing planned to properly confirm it works repeatedly. 

There's a rough plan to have time allocated to finish this to a state suitable for raising a PR in the next couple of weeks.

ma261065, 2025-08-03:
> I actually have a micropython USB Host integration underway already based on the TinyUSB Host support. The initial pr on this will support CDC serial, MSC (mass storage) and HID. There's future potential to expand this to runtime / arbitrary device support but the API / format of that will require far more significant planning.
> 
> I have this working on a branch already with all three classes mentioned above, however I think a little more fixup might still be required on cdc serial. Along with that I have a bunch of automated testing planned to properly confirm it works repeatedly.
> 
> There's a rough plan to have time allocated to finish this to a state suitable for raising a PR in the next couple of weeks.

This sounds amazing, so I thought I'd have a play with what you've already achieved. However, I'm obviously not doing something right when compiling it for ESP32.

First I cloned the `usbhost-clean` branch from you micropython fork. Then I #defined `MICROPY_HW_USB_HOST    (1)` in `~/micropython/ports/esp32/mpconfigport.h`

Then I compiled the firmware, which worked.... But the resultant firmware doesn't have the USBHost, only USBDevice.

It's probably some rookie error on my behalf, but would appreciate a nudge in the right direction... 

```
>>> import machine
>>> h = machine.
ADC             ADCBlock        DEEPSLEEP       DEEPSLEEP_RESET
EXT0_WAKE       EXT1_WAKE       HARD_RESET      I2C
I2S             PIN_WAKE        PWM             PWRON_RESET
Pin             RTC             SDCard          SLEEP
SOFT_RESET      SPI             Signal          SoftI2C
SoftSPI         TIMER_WAKE      TOUCHPAD_WAKE   Timer
TouchPad        UART            ULP_WAKE        USBDevice
WDT             WDT_RESET       bitstream       bootloader
deepsleep       dht_readinto    disable_irq     enable_irq
freq            idle            lightsleep      mem16
mem32           mem8            reset           reset_cause
sleep           soft_reset      time_pulse_us   unique_id
wake_reason

```

andrewleech, 2025-08-04:
It will probably need some more integration work to include it in esp32 port, particularly around the tinyusb config file. I was doing all my initial testing with rp2.

ma261065, 2025-12-09:
> There's a rough plan to have time allocated to finish this to a state suitable for raising a PR in the next couple of weeks.
> It will probably need some more integration work to include it in esp32 port, particularly around the tinyusb config file.

Have you managed to find any time to spend on this since? I'd love to explore this some more, but getting it to compile right now is beyond my ability - even after spending a lot of time on it.