micropython/micropython #17702: -fsanitizer=undefined diagnostics with longlong

jepler, 2025-07-17:
### Port, board and/or hardware

unix port, longlong variant

### MicroPython version

local version based on v1.25.0-387-g17fbc5abdc

### Reproduction

1. Modify `tools.ci.sh` to update the longlong build to include ubsan: 
    ```function ci_unix_longlong_build {
    ci_unix_build_helper VARIANT=longlong MICROPY_USE_COMPILER_PLUGIN=gcc \
    CFLAGS_EXTRA="-fsanitize=undefined -fno-sanitize=nonnull-attribute" LDFLAGS_EXTRA="-fsanitize=undefined -fno-sanitize=nonnull-attribute"
}
```
2. `(set -e; . tools/ci.sh; ci_unix_longlong_build; ci_unix_longlong_run_tests)`

### Expected behaviour

Tests succeed

### Observed behaviour

There are failures. Typical failure:
```
FAILURE /home/jepler/src/micropython/tests/results/basics_int_64_basics.py
--- /home/jepler/src/micropython/tests/results/basics_int_64_basics.py.exp	2025-07-17 12:00:04.723931234 -0500
+++ /home/jepler/src/micropython/tests/results/basics_int_64_basics.py.out	2025-07-17 12:00:04.727931280 -0500
@@ -117,6 +117,7 @@
 0
 0
 left shift negative
+../../py/objint_longlong.c:215:30: runtime error: left shift of negative value -10000000000000000
 -10000000000000000
 -10000000000000001
 -10000000000000002

```

```
4 tests failed: basics/int_64_basics.py micropython/viper_ptr32_store_boundary.py micropython/viper_ptr8_store_boundary.py micropython/viper_ptr16_store_boundary.py
```

### Additional Information

It might be a nice to add sanitizer builds of longlong (& nanbox?).

### Code of Conduct

Yes, I agree

jepler, 2025-07-22:
I'll try to do better at reporting which upstream commit I'm based on when I file issues.

projectgus, 2025-07-23:
@jepler Thanks! If it's a hassle then don't stress too much, better to have the bug report than not.

(BTW the m68k-micropython project is very neat, have been watching your progress on Mastodon!)

PR #17734 (2025-07-24): py/objint_longlong: Fix left shift of negative values.
### Summary

Closes #17702, thanks to @jepler for running these additional checks.

* The overflow check for 'longlong' representation left shift added in #16953 was wrong, and depended on UB. Rewrote to be the same as the small int left shift overflow check.
* Skip the viper "boundary" store tests if arbitrary integer support is disabled, these tests depend on generating very long integers and then truncating them.

Note: First version of this PR enabled UBSan for longlong tests, but this surfaced another issue so has been spun out to #17 [... cut]
closed 2025-07-24 by dpgeorge