micropython/micropython #17732: rp2: Can't send KeyboardInterrupt with os.dupterm()

sfe-SparkFro, 2025-07-21:
### Port, board and/or hardware

`rp2`

### MicroPython version

`MicroPython v1.25.0 on 2025-04-15; Raspberry Pi Pico 2 W with RP2350`

### Reproduction

Call `os.dupterm()` with UART or other compatible stream object:

```python
from machine import UART
import os
os.dupterm(UART(0), 0)
```

Run some code:

```python
import time
i = 0
while True:
    print(i)
    i += 1
    time.sleep(1)
```

### Expected behaviour

While code is running, pressing `Ctrl+C` in the USB REPL works as expected:

```
0
1
2
Traceback (most recent call last):
  File "<stdin>", line 6, in <module>
KeyboardInterrupt:
>>>
```

### Observed behaviour

While code is running, pressing `Ctrl+C` in the UART REPL fails to send the `KeyboardInterrupt` and halt execution.

Below, I pressed `Ctrl+C` a few times in the UART REPL, then once in the USB REPL. It then output extra `>>>`'s for each time I hit `Ctrl+C` in the UART REPL, indicating the `KeyboardInterrupt`'s were getting buffered or something?

```
0
1
2
3
Traceback (most recent call last):
  File "<stdin>", line 6, in <module>
KeyboardInterrupt:
>>>
>>>
>>>
>>>
>>>
```

### Additional Information

Although the provided replication code uses UART, this also happens with a Bluetooth REPL connection and is actually what we need to be fixed. This was just the simplest way to replicate.

### Code of Conduct

Yes, I agree

felixdoerre, 2025-07-25:
For interrupts to work, you need to set up your stream to receive interrupts. By default micropython only ready from the UART stream, when the repl is idle, and while the command is executing, no-one reads from the stream, so any received bytes are just buffered.

For webrepl there is a mechanism implemented to react to Ctrl+C: `dupterm_notify`, see here:

https://github.com/micropython/micropython-lib/blob/master/micropython/net/webrepl/webrepl.py#L134

So what I believe you need to do is to use that, and plug it into `UART.irq`, so a few bytes are read and buffered by the repl while the code is running.

sfe-SparkFro, 2025-07-30:
Thank you for the info! Is there any documentation for how to use `dupterm_notify`? Can't seem to find any info about it in the [MicroPython docs](https://docs.micropython.org/en/latest/search.html?q=dupterm_notify&check_keywords=yes&area=default) nor [this entire repo](https://github.com/search?q=repo%3Amicropython%2Fmicropython-lib%20dupterm_notify&type=code) (other than that example, which I'm not super clear on what exactly it's doing).

closed 2025-07-31 by sfe-SparkFro

robert-hh, 2025-07-31:
You could as well use a timer interrupt for dupterm_notify() with a short period.