micropython/micropython #17792: OSError 16 from time.sleep_ms()

victorallume, 2025-07-31:
### Port, board and/or hardware

stm32

### MicroPython version

MicroPython v1.25.0

### Reproduction

This is on a custom board. Code during the boot process calls `time.sleep_ms(1100)`
1. Hard reset the board
2. boot/main code starts running fine
3. Break out to a repl
4. (at this point, I can run time.sleep_ms() without problem)
5. soft reset the board
6. during the boot process, call to sleep_ms() fails

I cannot replicate the issue on v1.24.0 (i.e. something has changed going to v1.25.0)

### Observed behaviour

Call to time.sleep_ms() raises OSError 16 in the above conditions

### Additional Information

This is running on a custom board with an STM32H743 MCU.

### Code of Conduct

Yes, I agree

victorallume, 2025-07-31:
Was also unable to replicate on v1.24.1

victorallume, 2025-07-31:
Thanks, yes, I figure it's likely something in our code. However, given it only started happening with  v1.25, it seems that something has changed in micropython to manifest the issue. Unfortunately I don't have time to dig into it at the moment, so will stay with v1.24 until we need to upgrade

dpgeorge, 2025-08-04:
There were not many changes to the relevant code (wiznet, lwip, `ports/stm32/mpnetworkport.c`) between 1.24.1 and 1.25.0.  LwIP gained IPv6 support, do you enable IPv6?

> Call to time.sleep_ms() raises OSError 16

OSError 16 is EBUSY, which for stm32 most likely comes from the STM32 Cube HAL returning HAL_BUSY.  And that's most likely coming from `ports/stm32/spi.c:spi_transfer()`.  Maybe the wiznet driver is trying to send out data over SPI after a soft reset, and the SPI has been deinitialised by the soft reset.

How is the wiznet SPI configured for your application?

dpgeorge, 2025-08-04:
Oh, it's probably due to d42e39d87d37f20d74456222e603d0b6b409a556 and 5fa960c71912b0c5dace6ccd60a8dba48dc9e2da : SPI instances were deinitialised on soft reset.

Can you try either reverting that commit, or (better option) define `MICROPY_HW_SPI_IS_RESERVED` or `MICROPY_HW_SPI_IS_STATIC` in your board `mpconfigboard.h` file.

victorallume, 2025-08-05:
Thanks, that looks to be it. For the wiznet, I'm creating  SPI object in python than passing that the WIZNET5K initialiser. I had previously attempted to define the WIZNET config in mpboardconfig.h but couldn't get it working at the time so gave up and did it via python instead.

Defining `#define MICROPY_HW_SPI_IS_STATIC(id) ((id) == 1)` fixed the problem