micropython/micropython #17827: str.center() bug with unicode characters

faerot, 2025-08-04:
### Port, board and/or hardware

Raspberry Pico; Linux PC

### MicroPython version

MicroPython v1.25.0 on 2025-04-15; Raspberry Pi Pico W with RP2040
MicroPython v1.22.1+ds-1build2 on 2024-04-01; linux [GCC 13.2.0]

### Reproduction

Run this code in REPL or separately:
```print(len('°'.center(4)))```

### Expected behaviour

Expected to output 4

### Observed behaviour

Outputs 3

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

dlech, 2025-08-07:
It would be nice if we could create a label for issues related to unicode to make them easy to find.

There are a number of places MicroPython is still lacking unicode support, including this one.

Josverl, 2025-08-08:
@projectgus, see above request by David.
if I search open issues on UTF-8 or unicode I get 100 hits;  likely not all of them actually related, but having a label to tag them would be good. 
Id be happy to spend a few hours tagging the them.

<details><summary>UTF-8/unicode issues</summary>
<p>

below query using [github issue notebook](https://marketplace.visualstudio.com/items?itemName=ms-vscode.vscode-github-issue-notebooks) 

org:micropython is:open type:issue UTF-8 in:body OR org:micropython is:open type:issue UTF-8 in:title OR org:micropython is:open type:issue UTF-8 in:comments OR org:micropython is:open type:issue unicode in:body OR org:micropython is:open type:issue unicode in:title OR org:micropython is:open type:issue unicode in:comments

</p>
</details>

projectgus, 2025-08-11:
(and thanks for offering to tag related issues)