micropython/micropython #17828: STM32 , file system easy to corupt

quantrpeter, 2025-08-04:
### Port, board and/or hardware

Weact stm32 blackpill

### MicroPython version

Hi. I am using weact stm32 blackpill board. I can upload main.py and it will be run after reboot. After i unplug and plug in the USB around 1-3 times. All python files gone, i think the FS is corupted, it is too easy to damage. Can't build real product. Any workaround please?

### Reproduction

Hi. I am using weact stm32 blackpill board. I can upload main.py and it will be run after reboot. After i unplug and plug in the USB around 1-3 times. All python files gone, i think the FS is corupted, it is too easy to damage. Can't build real product. Any workaround please?

### Observed behaviour

WEACT blackpill board

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

robert-hh, 2025-08-04:
How do you upload the files? Do you use the PC file manager for it? Is the STM32 board mounted as driver to the PC? If yes, that's a configuration which is know to end up in file corruption. Better disable the external drive mode and use e.g. mpremote for file transfer. to disable the drive support, remove/comment the line with `pyb.usb_mode('VCP+MSC')` and add/enable the line with `pyb.usb_mode('VCP+HID')`.

quantrpeter, 2025-08-05:
i am using mpremote to upload all python, here is my script:

```
/Users/peter/workspace/at28-programmer-micropython>cat Makefile 
deploy:
	-mpremote rm :boot.py
	mpremote cp flashWrite.py :flashWrite.py
	mpremote cp flashRead.py :flashRead.py
	mpremote cp dummy.py :dummy.py
	mpremote cp main.py :main.py
	mpremote cp ssd1306.py :ssd1306.py
```

see this video, after i unplug and plug in again the usb, program gone.

https://github.com/user-attachments/assets/0ec9a7b9-1e3b-4898-8978-a35a3f0de9cd

I can ls the files but can't cat them, so i believe the FS is damaged

[image]

I designed a custom PCB to connect my weact blackpill stm32 to ssd1306 oled

[image]

[image]

@Josverl 

What firmware and version : latest source code from this repo
any specific changes to the firmware : no, i change none of any file
What file system are you using : I don't know, the default one?
how dit you upload main.py, using Mpremote, or some other way : mpremote
What host OS are you using in the repro : mac sonoma 14.6.1 (23G93)
how many is a few, : unplug and plugin the usb 1 - 3 times, then it is damaged
What else happens during plug/unplug? nothing else
are you using the blackpull as a removable drive in you host OS? no, but i see mac mounted it as a drive

thanks all

quantrpeter, 2025-08-05:
i discovered another thing, after i run mpremote to copy few files, i can't see it in the drive, in mac finder. But I can see them in "mpremote ls :" 

[image]

quantrpeter, 2025-08-05:
@robert-hh  Robert is super hero, after i added these on top of my main.py, the board won't be damaged. thanks all gentlemen

```
import pyb
pyb.usb_mode('VCP+HID')
```