micropython/micropython #17865: Uploading to esp32-s3 (4MB) corrupts file

gohai, 2025-08-08:
### Port, board and/or hardware

esp32-s3

### MicroPython version

MicroPython v1.26.0-preview.527.g593ae04ee on 2025-08-07; Generic ESP32S3 module with ESP32S3

mpremote 1.25.0, on macOS

### Reproduction

1. Upload a file to the board (e.g. `mpremote cp bar.py :/bar.py`)
2. Download the file again (`mpremote cp :/bar.py bar2.py`)
3. Compare the file content (`diff -u bar.py bar2.py`)

The files reproducibly differ for me. (Seeing syntax errors in repl as well, so the corruption doesn't seem to be caused by the second transfer back to the computer.)

### Expected behaviour

No file corruption. (Content should be identical.)

### Observed behaviour

File corruption, e.g.

```diff
 def create_wav_header(sampleRate, bitsPerSample, num_channels, num_samples):
     datasize = num_samples * num_channels * bitsPerSample // 8
     o = bytes("RIFF", "ascii")  # (4byte) Marks file as RIFF
     o += (datasize + 36).to_bytes(
         4, "little"
+    )  # (4byte) File size in bytes excluding this and Rittle"
     )  # (4byte) File size in bytes excluding this and RIFF marker
     o += bytes("WAVE", "ascii")  # (4byte) File type
     o += bytes("fmt ", "ascii")  # (4byte) Format Chunk Marker
     o += (16).to_bytes(4, "little")  # (4byte) Length of above format data
+    o += (1).to_bytes(2, "little")  # (2byte) Format tyat data
     o += (1).to_bytes(2, "little")  # (2byte) Format type (1 - PCM)
     o += (num_channels).to_bytes(2, "little")  # (2byte)
     o += (sampleRate).to_bytes(4, "little")  # (4byte)
     o += (sampleRate * num_channels * bitsPerSample // 8).to_bytes(4, "little")  # (4byte)
+    o += (num_channels * bitsPerSample /, "little")  # (4byte)
     o += (num_channels * bitsPerSample // 8).to_bytes(2, "little")  # (2byte)
     o += (bitsPerSample).to_bytes(2, "little")  # (2byte)
     o += bytes("data", "ascii")  # (4byte) Data Chunk Marker
     o += (datasize).to_bytes(4, "little")  # (4byte) Data size in bytes
     return o
```

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

robert-hh, 2025-08-08:
How should I read the file dump above? Are the green lines those in which the files differ? If yes, the a file corruption is highly unlikely. It then looks as if you compare different versions of the files, or a file was never written, or to different places..

gohai, 2025-08-08:
Sorry for not having been clear enough, @robert-hh!

The (green) lines are the output of diff-ing the _original_ file (as it was uploaded to the device) to the file as it was retrieved (downloaded) from it.

In other words: `mpremote cp` created a file on the device that had extra lines, which the file on the computer didn't have.

Hope this makes sense!

robert-hh, 2025-08-08:
That is a strange phenomenon. I tried to replicate your test with a S3 board, but could not. 
Does the upload or download of the file cause the corruption? You mentioned syntax errors, but you can as well just read the file and print the content in REPL to be sure.

 I use mpremote very often each day to copy files between board and PC, and never noticed any problem. So it may be something not related to mpremote. Sometimes I see echoes in the USB stack, which should not be there. That could cause such text duplication like you see. But that is hard to trace.

gohai, 2025-08-10:
I updated a ESP32 S2 board to v1.26 this morning, and saw exactly the same issue as I described (with an S3 board). Since this didn't happen to with v1.25 on that device, this could be a regression between the two versions.

agatti, 2025-08-11:
Just to be sure, does the `tusb_serial_device` example from ESP-IDF (in `examples/peripherals/usb/device/tusb_serial_device`) still exhibits the same issues on macOS?

Given the extreme difference in invalid behaviours for the same scenario (and now when entering text as well, which wasn't reported before), if Espressif's USB example works fine I'm starting to wonder if it's the current TinyUSB integration (either `ports/esp32/usb.[ch]`, `shared/tinyusb`, or both) that works *only* with the specific version that is optionally fetched as a submodule and needs changes for more recent versions.

agatti, 2025-08-11:
A friend of mine lent me their old macbook pro in order to see whether I could reproduce the issue.

On an ESP32S3-DevKitC-N8R16, building the latest `master` using ESP-IDF 5.4.2 (`BOARD=ESP32_GENERIC_S3 VARIANT=SPIRAM_OCT`) with those changes in `idf_component.yml` made the board crash after 35 back-to-back transfers (mac -> board, board -> mac, diff; so actually 70 USB operations) of a 250KBytes file.

I have to free some space on my desk to connect both my regular dev machine and this laptop to each USB-C port of the board to run this under JTAG and see exactly what is happening on the board.

I don't see any issues with the console though, no characters lost or repeated.  @robert-hh about the USB console echoes you reported, can you please give a bit more information on that?  Like MicroPython version, ESP-IDF version, platform you're connecting from, board details, etc.  Just to see if they're related to the commit in question or not.

agatti, 2025-08-13:
The plot thickens by *a lot* now - the issue is even getting board-dependent!?.

@bixb922 can you see the same REPL problems @gohai is having (hangs with `help()`)?

@gohai which ESP-IDF version did you use to rebuild the patched firmware?

agatti, 2025-08-13:
> > about the USB console echoes you reported, can you please give a bit more information on that? Like MicroPython version, ESP-IDF version,
> 
> At the moment I can reproduce it only with the ESP32C6 and ESP32C3 [...]

Thanks, @robert-hh the issue you're seeing is not related to this (neither the C6 nor the C3 have a full-featured USB interface).

gohai, 2025-08-13:
> [@gohai](https://github.com/gohai) which ESP-IDF version did you use to rebuild the patched firmware?

This was with a fresh checkout of `v5.4.2`.

gohai, 2025-08-19:
fcfc142524b9ea7137530b8c812a67ee35fcf086 fixes aforementioned repl hangs for me on the ESP32S3-S3-DEV-KIT-NXR8, @agatti. (I.e. all my issues are solved with that commit.) Thanks, and happy to test any future patches.

closed 2025-09-10 by dpgeorge