micropython/micropython #17871: Fail to flash the firmware due to the new I2C target mode feature

synodriver, 2025-08-08:
### Port, board and/or hardware

esp32 port, esp32-s3

### MicroPython version

master branch(593ae04), esp32-s3 board

### Reproduction

1. Build the firmware for esp32-s3, using ```make BOARD=ESP32_GENERIC_S3 BOARD_VARIANT=SPIRAM_OCT```
2. using ```mp-image-tool-esp32 --extract-app firmware.bin``` to get the app partition
3. using ```https://github.com/glenn20/micropython-esp32-ota``` to perform an ota update

### Expected behaviour

Expected to successfully perform an ota update. 

### Observed behaviour

The log shows 
```ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40383a26
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0xeb0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc1c
load:0x403cb700,len:0x2fbc
entry 0x403c88ac
E (773) i2c: CONFLICT! driver_ng is not allowed to be used with this old driver```

### Additional Information

Before the i2c slave mode feature was added, every thing works well. I tried to set ```CONFIG_I2C_SKIP_LEGACY_CONFLICT_CHECK=n``` but it didn't work.

### Code of Conduct

Yes, I agree

dxbsw, 2025-08-27:
I am using IDF v5.4.1 and encountered the same issue. I tried using `CONFIG_CODEC_I2C_BACKWARD_COMPATIBLE=y` to maintain compatibility with the legacy I2C driver, but it did not work.
Log output: `E (845) i2c: CONFLICT! driver_ng is not allowed to be used with this old driver.`

dxbsw, 2025-08-27:
> Use esp-idf v5.4.2  使用 esp-idf v5.4.2

Thank you very much, using idf v5.4.2 is no problem.

dpgeorge, 2025-08-28:
If you need to use IDF 5.4.1, the I2C target feature will need to be disabled, eg:
```
$ make CFLAGS_EXTRA=-DMICROPY_PY_MACHINE_I2C_TARGET=0
```

closed 2025-08-28 by dpgeorge