micropython/micropython #17896: @micropython.viper doesn't work on Alif.

kwagyeman, 2025-08-11:
### Port, board and/or hardware

OpenMV AE3, Alif E3

### MicroPython version

V1.26

### Reproduction

Any function with the viper decorator throws and error when being compiled:

```
MPY rpc.py
error compiling rpc.py:
Traceback (most recent call last):
  File "rpc.py", line 22, in rpc
SyntaxError: invalid arch
```

Something to do with this:

[image]

### Expected behaviour

Code compiles.

### Observed behaviour

Code doesn't compile.

### Additional Information

No, I've provided everything above.

### Code of Conduct

Yes, I agree

dpgeorge, 2025-08-12:
I did run native and viper and mpy tests for alif as part of the release testing, and they all passed.

Running them now again on OPENMV_AE3, and they still pass.  Eg:
```
$ ./run-tests -t a0 --via-mpy micropython/viper*.py
platform=alif arch=armv7emdp inlineasm=thumb
pass  micropython/viper_addr.py 
pass  micropython/viper_args.py 
pass  micropython/viper_binop_arith.py 
pass  micropython/viper_binop_arith_uint.py 
...
pass  micropython/viper_unop.py 
pass  micropython/viper_with.py 
39 tests performed (403 individual testcases)
39 tests passed
```

Can you please describe in more detail how you arrive at the error?  A minimal reproduction would be great.

(Maybe you forgot the `-march=armv7emdp` option?)

kwagyeman, 2025-08-12:
Mmm, maybe it's an issue with our build system in upstream OpenMV. I'll check and then close if that's the case.

kwagyeman, 2025-08-13:
It's because the MPY_CROSS_MCU_ARCH is not set. E.g. in lib/micropython/ports/stm32/stm32.mk:

```
MPY_CROSS_MCU_ARCH_f0 = armv6m
MPY_CROSS_MCU_ARCH_f4 = armv7m
MPY_CROSS_MCU_ARCH_f7 = armv7m
MPY_CROSS_MCU_ARCH_g0 = armv6m
MPY_CROSS_MCU_ARCH_g4 = armv7m
MPY_CROSS_MCU_ARCH_l0 = armv6m
MPY_CROSS_MCU_ARCH_l1 = armv7m
MPY_CROSS_MCU_ARCH_l4 = armv7m
MPY_CROSS_MCU_ARCH_h5 = armv7m
MPY_CROSS_MCU_ARCH_h7 = armv7m
MPY_CROSS_MCU_ARCH_n6 = armv7m # really armv8m
MPY_CROSS_MCU_ARCH_wb = armv7m
MPY_CROSS_MCU_ARCH_wl = armv7m
```

Which sets in the Makefile

```
# Options for mpy-cross
MPY_CROSS_FLAGS += -march=$(MPY_CROSS_MCU_ARCH_$(MCU_SERIES))
```

So, in lib/micropython/ports/alif/alif.mk

[image]

```
# Options for mpy-cross
MPY_CROSS_FLAGS += -march=armv7m # really armv8m
```

dpgeorge, 2025-08-13:
Ah, that makes sense.

See #17908 for a fix.

closed 2025-08-15 by kwagyeman